NLRWindSystems/WEIS #1: Streamline the compilation of TurbSim or pyTurbSim

johnjasa, 2020-08-24:
We cannot include XFoil directly in WEIS due to the GPL license.

nikhar-abbas, 2020-09-30:
This is a little bit out of my realm so I'm not sure exactly how this would work but there is a version of xfoil available on conda-forge.  I presume we might need to change some aspects of how we call xfoil, but it might be cleaner in the long run.

https://anaconda.org/conda-forge/libxfoil

Source code:
https://github.com/montagdude/libxfoil

gbarter, 2020-09-30:
That repo is a great find.  Here I had a whole detailed step-by-step guide on how to compile XFoil and write text-based input scripts when that repo is such a great find.

The only problem is the licensing.  As a GPL code, if we do anything that says `import xfoil` then our code will also have to be GPL according to the NREL lawyers.  They say that it has to be separately distributed, compiled, and executed via a system call (manual or automated).  I know the difference between the wrappers is semantics, but that is how they draw the line.

closed 2022-03-16 by gbarter