NLRWindSystems/WEIS #2: Determine which dependencies should be required

johnjasa, 2020-08-24:
Currently there are a huge number of dependencies due to the large number of projects installed as part of WEIS. Some of these may not be required for the core WEIS tools.

Resolution of this issue may require discussions about what should be considered core vs optional functionality of WEIS.

ptrbortolotti, 2020-08-27:
I'm listing here what I needed while setting up the examples:
1) ruamel_yaml - yaml management, possibly we can replace it with pyyaml that is already installed
2) jsonschema - validatation of the schemas, it's quite light we should be good adding it
3) xlrd - I believe it prints pandas dataframe to excel, it's used in either landbosse and/or orbit
4) petsc4py - parallelization scripts, only mac and ubuntu?
5) mpi4py - parallelization scripts, only mac and ubuntu?

gbarter, 2020-08-31:
will at least need the following conda packages: `numpy scipy pandas openmdao make cmake swig pytest geopy setuptools pip openpyxl`

And the following pip packages: `simpy marmot-agents`

closed 2020-09-02 by ptrbortolotti