NLRWindSystems/WEIS #14: Compile Issue for Windows environment

yonghoonlee, 2020-10-01:
Installing WEIS on Linux works fine, but I was not successful on Windows yet.
Followed the instruction on the README.md well, but getting following error message:

```
CMake suite maintained and supported by Kitware (kitware.com/cmake).
C:\Users\ylee196\Anaconda3\envs\weis-env\Library\bin\cmake.exe -S C:\Users\ylee196\Dropbox\WorkingFolder\WEIS\ROSCO -B build\temp.win-amd64-3.8\Release_rosco -DBUILD_SHARED_LIBS=OFF -DCMAKE_INSTALL_PREFIX=C:\Users\ylee196\Dropbox\WorkingFolder\WEIS\local -DCMAKE_WINDOWS_EXPORT_ALL_SYMBOLS=TRUE -G "MinGW Makefiles"
-- The Fortran compiler identification is GNU 5.3.0
-- Detecting Fortran compiler ABI info
-- Detecting Fortran compiler ABI info - done
-- Check for working Fortran compiler: C:/Users/ylee196/Anaconda3/envs/weis-env/Library/mingw-w64/bin/gfortran.exe - skipped
-- Checking whether C:/Users/ylee196/Anaconda3/envs/weis-env/Library/mingw-w64/bin/gfortran.exe supports Fortran 90
-- Checking whether C:/Users/ylee196/Anaconda3/envs/weis-env/Library/mingw-w64/bin/gfortran.exe supports Fortran 90 - yes
-- CMAKE_Fortran_COMPILER_ID = GNU
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/ylee196/Dropbox/WorkingFolder/WEIS/build/temp.win-amd64-3.8/Release_rosco
C:\Users\ylee196\Anaconda3\envs\weis-env\Library\bin\cmake.exe --build build\temp.win-amd64-3.8\Release_rosco -j 8 --target install --config Release
Scanning dependencies of target discon
[ 11%] Building Fortran object CMakeFiles/discon.dir/src/Constants.f90.obj
[ 22%] Building Fortran object CMakeFiles/discon.dir/src/ROSCO_Types.f90.obj
[ 33%] Building Fortran object CMakeFiles/discon.dir/src/Filters.f90.obj
[ 44%] Building Fortran object CMakeFiles/discon.dir/src/Functions.f90.obj
C:\Users\ylee196\Dropbox\WorkingFolder\WEIS\ROSCO\src\Functions.f90:412:30:

         CPfunction = saturate(CPfunction, 0.001, 1.0)
                              1
Error: Type mismatch in argument 'inputvalue' at (1); passed REAL(4) to REAL(8)
CMakeFiles\discon.dir\build.make:158: recipe for target 'CMakeFiles/discon.dir/src/Functions.f90.obj' failed
mingw32-make.exe[2]: *** [CMakeFiles/discon.dir/src/Functions.f90.obj] Error 1
CMakeFiles\Makefile2:93: recipe for target 'CMakeFiles/discon.dir/all' failed
mingw32-make.exe[1]: *** [CMakeFiles/discon.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
mingw32-make.exe: *** [all] Error 2
error: command 'C:\\Users\\ylee196\\Anaconda3\\envs\\weis-env\\Library\\bin\\cmake.exe' failed with exit status 2
```

gbarter, 2020-10-01:
This is an error with ROSCO and looks like one that can be easily fixed.  Even if this were fixed though, you will encounter bigger obstacles when it comes to compiling OpenFAST with the Anaconda toolchain.  That hasn't been worked out yet.  The OpenFAST team is getting close, as they already have a conda-installable version for linux/mac on their master branch, and they will be the ones to resolve the Windows issues.

gbarter, 2020-10-01:
@nikhar-abbas Might want to check out this error

nikhar-abbas, 2020-10-02:
Hi @yonghoonlee, @gbarter,
Thanks for pointing this out. 

Someone had similar issues compiling a recent version release on windows as well - they submitted a pull request with the fix here: https://github.com/NREL/ROSCO/pull/24.
@yonghoonlee, you should be able to edit cmakelists.txt with the correct changes to work through the bug you are running into, but, as Garrett said, you may run into problems further down the line.

Another note, a bug was accidentally introduced in the most recent version release of ROSCO, as pointed out in a few posts on the issues page for ROSCO. I’m working on getting those fixed and should have them ready to go by the end of the weekend. I do believe the version of ROSCO available through WEIS should be working fine, and I’ll be sure to submit the requisite pull requests once I have the new ROSCO versions sufficiently tested.

yonghoonlee, 2020-10-02:
@nikhar-abbas @gbarter Thank you for informative responses. Since OpenFAST compilation may not work with conda toolchain, I think it is better for me to just give up using Windows env. I mostly use Linux env, so it will not be a problem for me. Since the problem is already known and you are working on resolving this, I will close the issue. Thank you for your effort on WEIS project.

closed 2020-10-02 by yonghoonlee