NLRWindSystems/WEIS #15: WEIS/WISDEM not working

yonghoonlee, 2020-10-05:
WISDEM included in the WEIS is not recognized.

After installing WEIS using
`python setup.py develop`
WISDEM is not recognized:
```
python
import wisdem
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'wisdem'
```

I tried to install WISDEM separately as:
cd WEIS/WISDEM
`python setup.py develop`

and exhibited following error after several warnings:
```
collect2: error: ld returned 1 exit status
error: Command "/Users/yonghoonlee/opt/anaconda3/envs/weis-env/bin/x86_64-apple-darwin13.4.0-gfortran -Wall -g -Wall -g -undefined dynamic_lookup -bundle -Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs -Wl,-rpath,/Users/yonghoonlee/opt/anaconda3/envs/weis-env/lib -L/Users/yonghoonlee/opt/anaconda3/envs/weis-env/lib build/temp.macosx-10.9-x86_64-3.8/build/src.macosx-10.9-x86_64-3.8/wisdem/ccblade/_bemmodule.o build/temp.macosx-10.9-x86_64-3.8/build/src.macosx-10.9-x86_64-3.8/build/src.macosx-10.9-x86_64-3.8/wisdem/ccblade/fortranobject.o build/temp.macosx-10.9-x86_64-3.8/wisdem/ccblade/src/bem.o -L/Users/yonghoonlee/opt/anaconda3/envs/weis-env/lib/gcc/x86_64-apple-darwin13.4.0/7.3.0 -L/Users/yonghoonlee/opt/anaconda3/envs/weis-env/lib/gcc/x86_64-apple-darwin13.4.0/7.3.0/../../.. -L/Users/yonghoonlee/opt/anaconda3/envs/weis-env/lib/gcc/x86_64-apple-darwin13.4.0/7.3.0/../../.. -lgfortran -o wisdem/ccblade/_bem.cpython-38-darwin.so" failed with exit status 1
```

gbarter, 2020-10-05:
Not sure what the issue is there Yong Hoon.  I definitely have WEIS working via conda on my mac.  Can you reach out to me directly and maybe we can do a screen share?

yonghoonlee, 2020-10-08:
@gbarter Thank you for helping me resolve this issue.

For future reference, it apparently seems that the issue of incompatibility between macOS version and Conda-supported compilers suite version. The problem was observed after my computer was updated to the newest macOS version. Conda-supported compilers suite was remaining in the same version that I installed before. Thus, I ran
```
conda update compilers
```
and installed wisdem again. That resolved all the problem.

I will close the issue with this comment. Thank you.

closed 2020-10-08 by yonghoonlee