NLRWindSystems/WEIS #18: ROSCO Debug output shows some NaNs and Infitities with run_DLC.py test

yonghoonlee, 2020-10-17:
For DLC 6.1 and 6.3, some of the ROSCO debug output variables show NaNs and Infinities in RO.dbg files.
I am not sure if these are anticipated output (since they are extreme wind speed cases), or simulation failures.

gbarter, 2020-10-18:
@nikhar-abbas or @dzalkind or @ptrbortolotti may be able to help.

dzalkind, 2020-10-19:
Hi @yonghoonlee, I would expect to see some NaNs or Infinities in the RO.dbg files for parked cases because the wind speed estimator in ROSCO isn't set up to estimate the wind speed reliably in these cases.  Fortunately, it's not affecting the controller during these cases, so we'll probably leave this as it is in ROSCO until it does.

We use MPI to run on several nodes of a supercomputer, but locally you might want to try running with multiprocessing; an example is in `examples/run_general.py`

yonghoonlee, 2020-10-19:
@dzalkind Thank you for your answers. For us I believe MPI may be useful for our high performance computer system. However, I will also look into the run_general.py for the multiprocessing first, and close this issue. Thank you.

closed 2020-10-19 by yonghoonlee