NLRWindSystems/WEIS #27: LinearFAST.py, GenSpeed front factor of 0.95

yonghoonlee, 2020-11-14:
https://github.com/WISDEM/WEIS/blob/acb69e0c60c880a7193c56a78a2f42d5b5fd3056/weis/aeroelasticse/LinearFAST.py#L339
@dzalkind I am wondering why there is a front factor of 0.95 in the line linked above.
I tested with the full nonlinear simulation with the IEA-15MW turbine, and the RotSpeed and GenSpeed are exactly the same, as shown below:
[image]
Thank you.

dzalkind, 2020-11-16:
Hi Yong Hoon,

That `RefGenSpeed` is the generator speed set point used by the internal torque controller to find the steady state torque.  I set it that way because in near rated simulations, both the pitch and torque controllers may be active at the same time, so it's good to have the `RefGenSpeed` for the torque controller and pitch controller off set by a small amount so that the pitch and torque aren't trying to regulate the speed simultaneously; this can cause convergence issues in some cases.

Since the torque is still allowed to reach the full rated value and the pitch controller set point is at the full rated generator speed, this should have no effect on the linearization operating points.  It might be worth checking that the generator speed operating points in above rated wind speeds are at the desired rotational speed.  The outputs of the linearization runs can be viewed in the same way as nonlinear simulations using the `.out` or `.outb` files to help debug operating points, trim tolerances, and gains.

Best, Dan

yonghoonlee, 2020-11-25:
@dzalkind Thank you for your explanation. Now I get why you put value that is 5% smaller than the actual rated value.
Here, I have a few follow-up questions.
It seems that `VS_RtGnSp` is the "rated" generator speed, and the value you provided before was 7.56, but in your LinearFAST.py code, the `VS_RtGnSp` is overwritten with values dependent to the wind speed, meaning that:
```
VS_RtGnSp = 0.95 * [ROTOR SPEED] * GBRatio
```
So, I am a bit confused with this. When solving cases for wind speed ranging from 5m/s to 24m/s, rotor speed is also changing from 5.00 rpm to 7.56 rpm. I think "rated" generator speed is a turbine-specific parameter, and should be just a constant, but VS_RtGnSP is dependent to the rotor speed at each wind speed, making me a bit confusing.

Based on LinearFAST.py run, followings are the roughly calculated VS_RtGnSP values:

Wind speed = 5
Rotor speed = 5.00
VS_RtGnSp = 0.95 * 5.00 * 1 = 4.75

Wind speed = 6
Rotor speed = 5.00
VS_RtGnSp = 0.95 * 5.00 * 1 = 4.75

Wind speed = 7
Rotor speed = 5.03
VS_RtGnSp = 0.95 * 5.03 * 1 = 4.78

Wind speed = 8
Rotor speed = 5.73
VS_RtGnSp = 0.95 * 5.73 * 1 = 5.44

Wind speed = 9
Rotor speed = 6.41
VS_RtGnSp = 0.95 * 6.41 * 1 = 6.09

Wind speed = 10
Rotor speed = 6.52
VS_RtGnSp = 0.95 * 6.52 * 1 = 6.19

Wind speed = 11
Rotor speed = 6.56
VS_RtGnSp = 0.95 * 6.56 * 1 = 6.23

Wind speed = 12
Rotor speed = 7.56
VS_RtGnSp = 0.95 * 7.56 * 1 = 7.18

Wind speed = 13
Rotor speed = 7.57
VS_RtGnSp = 0.95 * 7.56 * 1 = 7.18

dzalkind, 2020-11-25:
Hey @yonghoonlee, this is a good catch. I think this should be fixed and we need to find a better way to figure out the rated generator speed value.   It's technically something we can change via control.  If you comment out [these lines](https://github.com/WISDEM/WEIS/blob/acb69e0c60c880a7193c56a78a2f42d5b5fd3056/weis/aeroelasticse/LinearFAST.py#L340) for now, it should work, but I will do some more testing to confirm.

yonghoonlee, 2021-07-22:
This issue is outdated, so I will close this issue.

closed 2021-07-22 by yonghoonlee