NLRWindSystems/WEIS #66: Modes for ElastoDyn

ptrbortolotti, 2021-01-19:
See previous discussions in 
https://github.com/OpenFAST/openfast/issues/631

Note that the issue has been temporarily fixed, but not fully resolved. 

Talking to Emmanuel, we foresee two paths forward:
1) Longer term, but cleaner - ElastoDyn computes the modes internally integrating the simplified calculations of BModes. Emmanuel would be happy to implement this. The code of BModes seems to have gone lost, but we could send some emails and see if we are lucky 
2) We refresh some old Python codes from Emmanuel that compute the modes. It is to be seen whether the modes match with the ones produced by BModes

We are open for feedbacks

gbarter, 2021-01-19:
I would be in favor of Option 1. The older BModes code is here:
https://github.com/old-NWTC/BModes
Incorporating this into ElastoDyn seems like a no-brainer to me.  This is not the source code for BModesJJ, which is what is needed for floating towers.  For that we would still have to use Frame3DD.

I don't think #2 is a viable option.  Finding the frequencies and modes in python, in addition to what Frame3DD does, is an unnecessary duplication of effort to accommodate the physics of ElastoDyn.

Maybe I'm missing something, but this seems low on the priority list to me.  Frame3DD is generating mode shapes that are compatible with ElastoDyn now.  Is it settled that the BModes-ElastoDyn combination is better than the Frame3DD-ElastoDyn combination?

ptrbortolotti, 2021-01-19:
I agree, option 1 is my favorite as well, but I thought it was harder. I'm surprised about the link above. I'll ask Emmanuel about it!

Regarding priority, deflections are now far closer than earlier thanks to your PR, although differences in deflections are still there

[image]
[image]

About theory, from JJ posts in the forum it seems that the "simplified" inputs to BModes are the correct ones to use with ED, while theory won't hold if we use frame3dd

ebranlard, 2021-01-20:
Hi both,

I guess the motivation for the second option is that we can implement a FEM in python with enough knobs to match Bmodes.  
Here's what I currently have [for instance](https://github.com/ebranlard/welib/blob/master/welib/FEM/fem_beam.py#L21). I have a test case for the tower modes [here](https://github.com/ebranlard/welib/blob/master/welib/FEM/tests/test_beam_linear_element.py) but inertia of the tower top would need to be implemented.
(This code is for a frame formulation that varies continuously along the element, I have element formulations for piecewise frame, and Timoshenko as well, just need to update it a bit. What's missing from this function is the centrifugal stiffness, which I believe can easily be computed by simply adding the "geometrical stiffness" Kg with an axial tension equal to the centrifugal loading. I have the element geometrical stiffness matrix implemented [here](https://github.com/ebranlard/welib/blob/master/welib/FEM/frame3dlin.py#L78) , so that shouldn't be too much work neither. ).

Option 1 is definitely nice because I believe a lot of users will like to have this option. We might need a user-input to defined the rotational speed used to tune the shape function (taking the initial RPM of the simulation might not be ideal).

If it were up to me, I would do both. Having the python version can be useful for other applications. (Also, I will inevitably implement it in Python one day or another... :))

ebranlard, 2021-01-20:
I've chatted a bit to Jason about option 1, and it seems that it wouldn't be the best to integrate the source code of BModes in OpenFAST since it's a complicated formulation. 

He's actually suggesting to obtain the polynomial coefficients using an FEM-approach where the deflections are assumed to be polynomials from the start. He will send us a paper on that. We can implement it in python and later in fortran maybe? 

That might give yet another shape function, but hopefully close to Bmode, and with time-series deflections that match..

gbarter, 2021-01-21:
Hi Emmanuel,

From a python point of view for WISDEM & WEIS, we are perfectly happy with pyFrame3DD.  It accomplishes everything on your list already.  The only hard part is we cannot "hack" the inputs as easily as BModes to comply with ED's assumptions.  It seems a little silly though to code up a new code just to play nice with ED.  If BModes cannot be merged in with ED, then perhaps we could use the SubDyn solver?  My understanding is that SubDyn can be used for the Tower in the new "TCF branch", but I likely don't have a full grasp of things here.  Or maybe just implement a new eigen-solver in ElastoDyn?  Is that what you were suggesting from Jason.

FYI: If you are moving into a higher-order polynomial basis for FEM, you could consider pBeam.  That is not the same as assuming polynomial deflections, but it is another good starting point.

closed 2021-03-16 by ptrbortolotti