NLRWindSystems/WEIS #68: Anaconda openmpi petsc4py version issue on mac

yonghoonlee, 2021-01-21:
The most recent version of petsc4py is compatible only up to openmpi 4.0.x while anaconda version of openmpi is currently 4.1.x. Due to this issue WEIS did not run correctly on the recently-updated conda environment.

Resolution:

```
conda install -y openmpi=4.0
```
will resolve the issue, and I believe this will not be a problem later, when petsc4py is updated to have compatibility with openmpi 4.1.x.

gbarter, 2021-01-21:
Hi Yong Hoon,

We have noticed this issue too and been passing around the solution via word-of-mouth.  Thank you for formally documenting it here.  Hopefully conda forge will sort out this inconsistency soon.

yonghoonlee, 2021-07-22:
This issue no longer persists. I will close this issue.

closed 2021-07-22 by yonghoonlee