NLRWindSystems/WEIS #69: Linearization capability is broken in the current WEIS version.

yonghoonlee, 2021-01-22:
When I try to run LinearFAST or my own linearization code, OpenFAST linearization exhibit segmentation faults with ????? messages. Since I have not been updating WEIS for long time, and recently updated, I do not know when it was broken. Would you be able to check if something is wrong in LinearFAST, @dzalkind ? Steady state calculation works well, but when it comes to the linearization stage, after TRIM condition reaches, segmentation fault occurs. Thank you.

yonghoonlee, 2021-01-22:
by the way, I was working on the [develop](https://github.com/WISDEM/WEIS/tree/develop) branch.

dzalkind, 2021-01-22:
Hi @yonghoonlee,

There was an issue in one of the latest versions of openfast (https://github.com/OpenFAST/openfast/issues/620) that has since been resolved.  I can look at this shortly and make sure we are using a working version here in WEIS.  In the meantime, pointing to the dev branch (https://github.com/OpenFAST/openfast/tree/dev) of openfast should work.

yonghoonlee, 2021-01-22:
@dzalkind 
Thank you for quick response.
I was also going to ask about the tip deflection incompatibility, and it seems that this was also discussed in [OpenFAST/openfast#620](https://github.com/OpenFAST/openfast/issues/620) as well. I will try with the most recent dev version from OpenFAST repository [https://github.com/OpenFAST/openfast/tree/dev](https://github.com/OpenFAST/openfast/tree/dev). Thank you for your help.

dzalkind, 2021-01-22:
No problem! In a way, openfast ignoring the tip deflection showed me that we don't really need to be solving for the steady state solutions before linearizing.  So we can skip those parts in LinearFAST.  I was able to do that in my fork [here](https://github.com/dzalkind/WEIS/blob/b9e16a459d4b368de4833fd71cf142509c354b35/weis/aeroelasticse/LinearFAST.py#L421) and should have a PR coming soon with an updated LinearFAST class before our interim release at the end of the month.

yonghoonlee, 2021-01-23:
@dzalkind 
It seems that the latest OpenFAST dev branch has changed its input structures in AeroDyn15. I tried to bypass this, but a lot of tools included in the WEIS is reverting my changes automatically, so I was not successful running with the latest dev branch of OpenFAST for linearization.
I hope things included in the WEIS can be fixed soon, so that we can run some linearization studies using WEIS. Thanks.

yonghoonlee, 2021-07-22:
This issue is outdated, so I will close this issue.

closed 2021-07-22 by yonghoonlee