NLRWindSystems/WEIS #91: TStart!=0 in OpenFAST introduces bugs into stats

ptrbortolotti, 2021-03-16:
## Description
The Python call to OpenFAST generate statistics that are wrong when TStart!=0

### Steps to reproduce issue

1. Start from example 03 https://github.com/WISDEM/WEIS/tree/develop/examples/03_NREL5MW_OC3_spar
2. Reduce the wind speeds to a single one (I kept 5 m/s) for faster debugging https://github.com/WISDEM/WEIS/blob/6a8861f2fc830d77536aaa4224deb8192eb681cd/examples/03_NREL5MW_OC3_spar/modeling_options.yaml#L92
3. Check the stats generated at this line https://github.com/WISDEM/WEIS/blob/6a8861f2fc830d77536aaa4224deb8192eb681cd/weis/aeroelasticse/runFAST_pywrapper.py#L259 by printing ss['NREL5MW_OC3_spar_powercurve_0']['RtAeroCp']. You will obtain
{'abs': 75.4937515258789, 'integrated': -257.78151857376105, 'max': 0.0, 'mean': -17.638143743311133, 'min': -75.4937515258789, 'std': 20.531107334656}
4. Check in the .outb file or in the figure below that the max of RtAeroCp is not 0 and the mean is not -17

[image]

### Current behavior
The stats are wrong. My suspect is that between Time=0 and Time=TStart all channels are set to 0 and they are included in the statistics, while they should be left out.

### Expected behavior
The stats should only include from TStart to the end of the time history

### Code versions
Latest WEIS develop

ptrbortolotti, 2021-03-16:
The problem might hide behind the fact that we are not passing the trim_data tuple
https://github.com/WISDEM/WEIS/blob/6a8861f2fc830d77536aaa4224deb8192eb681cd/pCrunch/pCrunch/analysis.py#L39

ptrbortolotti, 2021-03-16:
yep that's the problem, although I'm not sure how to best pass it, since the class is initialized here
https://github.com/WISDEM/WEIS/blob/6a8861f2fc830d77536aaa4224deb8192eb681cd/weis/aeroelasticse/runFAST_pywrapper.py#L87

I'm happy to hear suggestions, probably @JakeNunemaker will know best

gbarter, 2021-03-16:
@ptrbortolotti : Thank you for tracking this down.  This one must have taken some digging.

There will have to be changes on both the pCrunch side and the AeroelasticSE side.  

@JakeNunemaker : Can we take the trim_outputs argument out of the constructor and put it as an argument into this function:
https://github.com/WISDEM/WEIS/blob/develop/pCrunch/pCrunch/analysis.py#L119

The start/stop time values could easily change case by case in a batch run, so it would be good to let them vary.  If you do that, then I think we only need to add them here:
https://github.com/WISDEM/WEIS/blob/develop/weis/aeroelasticse/runFAST_pywrapper.py#L174
and then everything else should fall in line.

JakeNunemaker, 2021-03-16:
You can actually pass the trim data tuple into the initialization of the Output class on the line above:

https://github.com/WISDEM/WEIS/blob/6a8861f2fc830d77536aaa4224deb8192eb681cd/weis/aeroelasticse/runFAST_pywrapper.py#L173

I think that resolves your first point @gbarter. As for varying by case by case, the `execute` command is ran on a case by case basis too, right? So we would have to wire it up through that function into the the line referenced above?

ptrbortolotti, 2021-03-16:
we currently have a single TStart and a single TMax among the modeling options, so one value could be enough for now. Let me see if I can fix things as you suggest @JakeNunemaker

gbarter, 2021-03-16:
I see, so we just add a line of, `output.trim_data(start, stop)` in the `execute()` function and we should be good to go?

ptrbortolotti, 2021-03-16:
this should fix things 
https://github.com/WISDEM/WEIS/pull/92

closed 2021-03-31 by ptrbortolotti