NLRWindSystems/WEIS #130: GPL-v3 module usable?

yonghoonlee, 2021-07-09:
WEIS is licensed under Apache license and is not compatible to include GPL-v3 code inside it.
However, in my understanding, just importing modules developed under GPL-v3 does not change WEIS itself.
So, I would like to see if using python module that is developed under GPL-v3 is possible under its license.
I would be more comfortable using LGPL, but GPL is more trickier.

gbarter, 2021-07-09:
Hi Yonghoon,

This is a good question.  We have had numerous, lengthy conversations with the NREL software and legal teams about open source licenses over the past few years.  At different times we have also wanted to include GPL packages in WISDEM or WEIS, including XFoil and OpenSCAD.  We reached an agreement that if the user is responsible for downloading and installing the GPL code, then it can be automatically executed in our software with a system-call via a user-specified path.  If you want to do an "import gpl-package-foo" or link a library at compile time, then there are more serious legal issues that NREL has asked us to keep clear of.

That being said, I don't think we've done our due-diligence on secondary or tertiary dependencies for our codes.  What package did you have in mind?

Cheers,
Garrett

yonghoonlee, 2021-07-09:
I was looking for a good QP solver for Python-DTQP implementation from CSU. I was looking at the CVXOPT, an LP/QP solver, which is under GPL-v3, and thought it could be risky to use that code due to licensing issue. I will give this issue link to CSU team as well so they can aware of this issue when choosing the QP solver.

gbarter, 2021-07-09:
@johnjasa or @dzalkind : Have you used a python QP solver before?

A quick google search gives me:
https://pypi.org/project/quadprog/ (GPLv2)
https://github.com/coin-or/qpOASES (LGPL)
http://cvxopt.org/index.html (GPLv3)
https://github.com/liuq/QuadProgpp/ (MIT)
https://pypi.org/project/mystic/ (BSD)

Other choices for sparse solvers and some commercial options:
https://github.com/stephane-caron/qpsolvers

SLSQP is also ubiquitous in many packages

AthulKrishnaSundarrajan, 2021-07-12:
I went through the WEIS tool, and I noticed that the tool 'pyoptsparse' is licensed under the LGPL license.
Does this mean that tools like qpOASES (LGPL) be used along with the apache license that WEIS has?

Best,
Athul

gbarter, 2021-07-12:
As I understand it from the NREL software/legal team, LGPL is manageable since it doesn't have the viral copy-left terms like GPL does, but I haven't read LGPL terms closely myself.

yonghoonlee, 2021-07-12:
@AthulKrishnaSundarrajan 
If something is in LGPL, it is less problematic since LGPL stands for "Lesser" GPL, which allows it to be included as-is in a form of library, and does not affect the mother software's license.

yonghoonlee, 2021-07-22:
I think this issue is well answered, so I will close the issue.

closed 2021-07-22 by yonghoonlee