NLRWindSystems/WEIS #132: OpenFAST Python Library Error Reporting

dzalkind, 2021-07-12:
## Description
When using the OpenFAST Python library, some errors are not reported.  

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Set `PtfmMass_Init: 0` 
2. Set `volume: 1e3`(https://github.com/WISDEM/WEIS/blob/1e4dbf6728050f75cee08cd483fe57c5614488fe/examples/03_NREL5MW_OC3_spar/modeling_options.yaml#L69) (https://github.com/WISDEM/WEIS/blob/1e4dbf6728050f75cee08cd483fe57c5614488fe/examples/03_NREL5MW_OC3_spar/nrel5mw-spar_oc3.yaml#L345)
3. Run `python examples/03_NREL5MW_OC3_spar/weis_driver.py`

### Current behavior
The FOWT sinks and the OpenFAST library returns `Error 0: b''`

### Expected behavior
If you use the executable: (set via the modeling options)
```
openfast:
    file_management:
        FAST_namingOut:    NREL5MW_OC3_spar         # Name of the OpenFAST output files
        FAST_runDirectory: fast_outputs         # Path to folder with the OpenFAST output files
        use_exe: True
```

OpenFAST returns:
```

 FAST_Solution:FAST_AdvanceStates:SolveOption2c_Inp2AD_SrvD:InflowWind_CalcOutput:CalcOutput:IfW_FF
 Wind_CalcOutput [position=(0.65225, -6.76065E-02, 0.98863)  in wind-file coordinates]: FF wind
 array boundaries violated. Grid too small in Z direction (height (Z=0.98863 m) is below the grid
 and no tower points are defined).

```

gbarter, 2021-07-12:
Thanks for raising this.  I did spend some time recently poking around that available strings in the OpenFAST python library wrapper and could not find the error message anywhere (which is why you get an empty byte-character string).  This is something @rafmudaf will know more about.

dzalkind, 2021-07-12:
Hey @rafmudaf, I found another possible library issue:

I was trying to make sure the library call was covered and changed this test on my own fork: `https://github.com/dzalkind/WEIS/tree/global_dlc_driver/weis/aeroelasticse/test/test_OF_utils.py`

I see:

```
======================================================================
ERROR: testWrapper (__main__.TestOFutils)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/dzalkind/Tools/WEIS-1/weis/aeroelasticse/test/test_OF_utils.py", line 65, in testWrapper
    fast_wr.execute()
  File "/Users/dzalkind/Tools/WEIS-1/weis/aeroelasticse/runFAST_pywrapper.py", line 174, in execute
    openfastlib = FastLibAPI(self.FAST_lib, os.path.abspath(os.path.basename(writer.FAST_InputFileOut)))
  File "/Users/dzalkind/Tools/WEIS-1/OpenFAST/glue-codes/python/openfast_library.py", line 22, in __init__
    self._initialize_routines()
  File "/Users/dzalkind/Tools/WEIS-1/OpenFAST/glue-codes/python/openfast_library.py", line 50, in _initialize_routines
    self.FAST_AllocateTurbines.argtypes = [
  File "/Users/dzalkind/opt/anaconda3/envs/weis-env1/lib/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/Users/dzalkind/opt/anaconda3/envs/weis-env1/lib/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: dlsym(RTLD_DEFAULT, FAST_AllocateTurbines): symbol not found
```

If this line gets set to the binary, OpenFAST runs: https://github.com/dzalkind/WEIS/blob/7613c82f16dd18f97985883dbe1b5d223c50a55d/weis/aeroelasticse/test/test_OF_utils.py#L60

rafmudaf, 2021-07-13:
All right I believe I've found it. Here's the error message through the Python interface:
```
  Total Real Time:       0.56503 minutes
  Total CPU Time:        2.9664 minutes
  Simulation CPU Time:   2.325 minutes
  Simulated Time:        4.79167E-02 minutes
  Time Ratio (Sim/CPU):  2.06096E-02
Nonlinear constraints
{'aeroelastic.Max_PtfmPitch': array([0.]),
 'aeroelastic.rotor_overspeed': array([0.]),
 'sse_tune.tune_rosco.PC_Ki': array([-0.00233395]),
 'sse_tune.tune_rosco.PC_Kp': array([-0.00874691])}

Linear constraints
None

Objectives
{'wisdem.wt.floatingse.load.plat.platform_mass': array([9915912.91624534])}

Traceback (most recent call last):
  File "/home/miniconda3/envs/weis/lib/python3.8/site-packages/openmdao/core/system.py", line 2220, in _call_user_function
    yield
  File "/home/miniconda3/envs/weis/lib/python3.8/site-packages/openmdao/core/explicitcomponent.py", line 245, in _compute_wrapper
    self.compute(*args)
  File "/home/WEIS/weis/aeroelasticse/openmdao_openfast.py", line 483, in compute
    summary_stats, extreme_table, DELs, case_list, case_name_list, dlc_list  = self.run_FAST(inputs, discrete_inputs, fst_vt)
  File "/home/WEIS/weis/aeroelasticse/openmdao_openfast.py", line 1446, in run_FAST
    summary_stats, extreme_table, DELs, chan_time = fastBatch.run_serial()
  File "/home/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 294, in run_serial
    _name, _ss, _et, _dl, _ct = evaluate(c)
  File "/home/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 417, in evaluate
    FAST_Output = fast.execute()
  File "/home/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 176, in execute
    openfastlib.fast_run()
  File "/home/WEIS/OpenFAST/glue-codes/python/openfast_library.py", line 208, in fast_run
    self.fast_sim()
  File "/home/WEIS/OpenFAST/glue-codes/python/openfast_library.py", line 178, in fast_sim
    raise RuntimeError(f"Error {_error_status.value}: {_error_message.value}")
RuntimeError: Error 4: b'FAST_Solution:FAST_AdvanceStates:SolveOption2c_Inp2AD_SrvD:InflowWind_CalcOutput:CalcOutput:IfW_FFWind_CalcOutput [position=(0.65225, -6.76065E-02, 0.98863)  in wind-file coordinates]: FF wind array boundaries violated. Grid too small in Z direction (height (Z=0.98863 m) is below the grid and no tower points are defined).'
```

rafmudaf, 2021-07-13:
> I was trying to make sure the library call was covered

@dzalkind I'm going to work on getting this covered in the tests.

rafmudaf, 2021-07-13:
@dzalkind I believe #135 and #134 resolve this issue. What do you think?

dzalkind, 2021-07-13:
Yep, thanks @rafmudaf

closed 2021-07-13 by dzalkind