NLRWindSystems/WEIS #153: OpenFAST Library Fails on Eagle

dzalkind, 2021-11-09:
## Description
When running WEIS with the OpenFAST library, there is an MKL error.  More details on this issue can be found in #152 .

A workaround is to use the executable by setting the following modeling options: 
```
General:
    openfast_configuration:
        use_exe: True
```

gbarter, 2021-11-12:
I Google around and tried a couple of things and succeeded in fixing this one.  I'm still not sure what changed that things all of a sudden stopped working.

@rafmudaf : This required loading the `mkl_rt` library before any of the others.  I editted the bottom of the `FindMKL.cmake` file to be:

```
find_library(MKL_RT_LIB
  NAMES mkl_rt libmkl_rt mkl_rt_dll
  PATHS ${MKLSEARCHPATHS}
  NO_DEFAULT_PATH)

if (MKL_IFACE_LIB AND MKL_SEQ_LIB AND MKL_CORE_LIB AND MKL_RT_LIB)
  set(MKL_LIBRARIES ${MKL_RT_LIB} ${MKL_IFACE_LIB} ${MKL_SEQ_LIB} ${MKL_CORE_LIB})
else()
  set(MKL_LIBRARIES "")
  set(MKL_INCLUDE_DIRS "")
endif()

include(FindPackageHandleStandardArgs)
find_package_handle_standard_args(MKL DEFAULT_MSG
  MKL_LIBRARIES MKL_RT_LIB MKL_IFACE_LIB MKL_SEQ_LIB MKL_CORE_LIB) # MKL_INCLUDE_DIRS)
mark_as_advanced(
  MKL_INCLUDE_DIRS MKL_LIBRARIES MKL_RT_LIB MKL_IFACE_LIB MKL_SEQ_LIB MKL_CORE_LIB)
```

Is this something you want to explore or should I leap to an OpenFAST PR?  I feel like at some point you and I might have discussed this `mkl_rt` library before?

dzalkind, 2021-11-16:
Nice find...thanks Garrett!

dzalkind, 2022-01-28:
Hey @gbarter, I'm getting this error again when running HAMS on eagle.  Do you have any ideas for a workaround or fix?  I'm not seeing anything similar in pyHAMS to the above solution, but I am a cmake noob :-)

I'm on the [level1 branch](https://github.com/WISDEM/WEIS/tree/level1) with a fresh install on eagle. I merged develop into that branch earlier today. Running [this example](https://github.com/WISDEM/WEIS/blob/level1/examples/03_NREL5MW_OC3_spar/weis_freq_driver.py) will reproduce the issue.

gbarter, 2022-01-29:
@dzalkind The earlier fix was in the compilation setup for OpenFAST.  The error you are seeing now is for pyHAMS.  I replicated the fix and that example is now working.

closed 2022-08-16 by dzalkind