NLRWindSystems/WEIS #165: mpirun

nickjohnson13, 2022-02-18:
I'm getting an error when I use the mpirun command from the latest WEIS/dev build when running WISDEM/examples/03_blade:

$ mpirun -np 4 python blade_driver.py

Traceback (most recent call last):
  File "blade_driver.py", line 15, in <module>
    wt_opt, modeling_options, analysis_options = run_wisdem(fname_wt_input, fname_modeling_options, fname_analysis_options_no_opt)
  File "/Users/njohnso1/Documents/Projects/WEIS_dev/WISDEM/wisdem/glue_code/runWISDEM.py", line 41, in run_wisdem
    raise ValueError("ERROR: please reduce the number of cores, currently set to " +
ValueError: ERROR: please reduce the number of cores, currently set to 4, to the number of finite differences 0, which is equal to the number of design variables DV for forward differencing and DV times 2 for central differencing, or the parallelization logic will not work

Thanks,
Nick

ptrbortolotti, 2022-03-16:
My apologies for the slow reply. The code errors out because this call (https://github.com/WISDEM/WEIS/blob/1db6a7e0921a55f074a16221c672de898155c0cf/WISDEM/examples/03_blade/blade_driver.py#L15) to WISDEM has no design variables associated to it (no optimization is set here https://github.com/WISDEM/WEIS/blob/master/WISDEM/examples/03_blade/analysis_options_no_opt.yaml), so there is only a single call to WISDEM and the parallelization cannot happen.

ptrbortolotti, 2022-03-16:
commenting out this line https://github.com/WISDEM/WEIS/blob/1db6a7e0921a55f074a16221c672de898155c0cf/WISDEM/examples/03_blade/blade_driver.py#L15 will make the mpi call work

15280560196, 2024-06-26:
I also have the same problem, check whether the source code is wrong?

15280560196, 2024-06-26:
[image]