NLRWindSystems/WEIS #171: Install issue

SeongwooCheon, 2022-04-25:
Hi,

When I install the WEIS following the guide, the error is occured like this.

```
I typed this command
conda install -y cmake cython control dill git jsonschema make matplotlib-base numpy openmdao openpyxl pandas pip pyoptsparse pytest python-benedict pyyaml ruamel_yaml scipy setuptools simpy slycot smt sortedcontainers swig
```

the error is 

```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - pyoptsparse

Current channels:

  - https://conda.anaconda.org/conda-forge/win-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/win-64
  - https://repo.anaconda.com/pkgs/free/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

How can I solve this?

gbarter, 2022-04-25:
The `pyoptsparse` package is not available on Windows (see its Anaconda.org [page](https://anaconda.org/conda-forge/pyoptsparse)).  Beyond `pyoptsparse`, the conda-supplied compilers also are not capable of compiling OpenFAST on Windows.  You would have to use the OpenFAST release binaries for Windows and hack the WEIS setup a bit.  For now, we recommend using the Windows Ubuntu Subsystem if you are on a Windows machine.

SeongwooCheon, 2022-04-25:
I want to transform WISDEM output into OpenFAST input. Is there Python or FORTRAN code like this?
[https://github.com/OpenFAST/python-toolbox/blob/dev/pyFAST/converters/openfastToHawc2.py](url)

SeongwooCheon, 2022-05-10:
thank you for your response:)

closed 2022-05-10 by SeongwooCheon