NLRWindSystems/WEIS #183: Automatic writing of various DLC input files

kpramng, 2022-08-09:
# Description of feature

Already, Is there any Python based script available that can be used to run a set of DLC's at one shot. For example, a script that can run NTM simulations from 3m/s to 25 m/s (basically create the input files for such a simulation) along with the cases like idling, idling above cut out etc.

# Potential solution
A Python script that can create multiple input files or point to various files like controllers, brakes etc. for each of the simulations.

Best,
Karthik

dzalkind, 2022-08-09:
Hi Karthik,

Yes, in the `develop` branch, [this script](https://github.com/WISDEM/WEIS/blob/develop/examples/02_control_opt/weis_driver_loads.py) will run [these DLCs](https://github.com/WISDEM/WEIS/blob/fefe87833d77366f204bf58fa3a80d55f594ef2c/examples/02_control_opt/modeling_options_loads.yaml#L111) for [this OpenFAST model.](https://github.com/WISDEM/WEIS/blob/fefe87833d77366f204bf58fa3a80d55f594ef2c/examples/02_control_opt/modeling_options_loads.yaml#L31)

If you uncomment [this line](https://github.com/WISDEM/WEIS/blob/fefe87833d77366f204bf58fa3a80d55f594ef2c/examples/02_control_opt/modeling_options_loads.yaml#L125) it should run DLC 1.1 from cut-in to cut-out, with 6 seeds each.  You can also select certain wind speeds and number of seeds. 

I hope this helps.

Best, Dan

kpramng, 2022-08-09:
Hi Dan,

Thanks a lot for getting back, this really helps. I am not that great with programming and hence pardon me for my naïve questions.

The *.yaml files, are they Python scripts that I can run in Anaconda etc. and can I use the script to run a Open fast model that I have.

Best,
Karthik

dzalkind, 2022-08-09:
Hi Karthic,

You run python scripts with `.py` extensions in an Anaconda environment using `python weis_driver_loads.py`, for example.

The `.yaml` files are inputs to WEIS.  If you look in `weis_driver_loads.py`, you will see they are set as the geometry, modeling, and analysis inputs to WEIS.

kpramng, 2022-08-09:
Hi Dan,

So, can I run the python script 'weis_driver_loads.py' in Python as is separately & if so is it so that I have to point to the path of .yaml file in the Python script.

Again, excuse me for my naive queries in coding.

Best,
Karthik

dzalkind, 2022-08-09:
Yes, right around here: https://github.com/WISDEM/WEIS/blob/fefe87833d77366f204bf58fa3a80d55f594ef2c/examples/02_control_opt/weis_driver_loads.py#L9

kpramng, 2022-08-09:
Thanks a lot Dan. I will git a shot and update.

Best,
Karthik

closed 2022-08-16 by dzalkind

kpramng, 2022-09-06:
> Hi Karthik,
>
> Yes, in the `develop` branch, [this script](https://github.com/WISDEM/WEIS/blob/develop/examples/02_control_opt/weis_driver_loads.py) will run [these [...]

Hi Dan,

Do I need to install WEIS to run this script for a Onshore Wind turbine. Is there a similar file for Onshore turbines.

Regards,
Karthik

kpramng, 2022-09-26:
Hi Dan,

Thanks a lot for the reply. Is there a way to create one .fst file that will have the wind speeds from cut-in to cut-out or separete .fst files are needed for running simulations from cut-in to cut-out. For example from 3m/s to 25 m/s, do I need to create 23 .fst files?

Best regards,
Karthik

kpramng, 2022-10-03:
Hi Dan,

Probably a dumb question. If I want to run  [https://github.com/WISDEM/WEIS/blob/develop/examples/02_control_opt/modeling_options_loads.yaml](url), does that mean that I have to run WEIS so that it generates all the .fst files and the corresponding inputs, for example like. -dat files etc. needed to run an OpenFast Aeroelastic simulation? If so, can you help me find the instructions on how to run WEIS so that those inputs will be created.

Best regards,
Karthik

dzalkind, 2022-10-03:
Yes, you will need to run WEIS.  I added some documentation here:

https://github.com/WISDEM/WEIS/tree/develop/examples/02_control_opt

Running the weis_driver.py files will load modeling, geometry, and analysis options to run WEIS.

More info can be found here: https://github.com/WISDEM/WEIS/blob/develop/docs/how_weis_works.rst

kpramng, 2022-10-04:
Hi @dzalkind ,

I got the below error while trying to run WEIS with the command python weis_driver.py

When I first installed WEIS, I used the instructions mentioned here  https://github.com/WISDEM/WEIS

[image]

gbarter, 2022-10-04:
It looks like WEIS isn't installed in the `weis-env` environment.  Make sure you have installed WEIS there by doing `python setup.py develop` in your `Users\prakakar\source\WEIS` directory.  It is also a good idea to do a `git pull` before that to ensure you are current with the repository.

kpramng, 2022-10-04:
I will try that out. Should I clone the develop branch or the main would suffice?

gbarter, 2022-10-04:
Aim for the `develop` branch, as that will be released into `main` shortly.

kpramng, 2022-10-04:
I followed the instructions [https://github.com/WISDEM/WEIS/tree/develop](url) 
But the same error is popping up 

(base) C:\>cd C:\Users\prakakar\source\WIES_Develop\WEIS\examples\02_control_opt

(base) C:\Users\prakakar\source\WIES_Develop\WEIS\examples\02_control_opt>conda activate weis-env

(weis-env) C:\Users\prakakar\source\WIES_Develop\WEIS\examples\02_control_opt>python weis_driver.py
Traceback (most recent call last):
  File "weis_driver.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
ModuleNotFoundError: No module named 'weis'

(weis-env) C:\Users\prakakar\source\WIES_Develop\WEIS\examples\02_control_opt>python weis_driver.py
Traceback (most recent call last):
  File "weis_driver.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
ModuleNotFoundError: No module named 'weis'

(weis-env) C:\Users\prakakar\source\WIES_Develop\WEIS\examples\02_control_opt>python weis_driver.py
Traceback (most recent call last):
  File "weis_driver.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
ModuleNotFoundError: No module named 'weis'

(weis-env) C:\Users\prakakar\source\WIES_Develop\WEIS\examples\02_control_opt>

gbarter, 2022-10-04:
The installation is failing in some way.  Please post the results of the install instructions you followed.

Also note that WEIS does not run on Windows unless you are running in the Windows Subsystem Linux

kpramng, 2022-10-04:
I don't have Linux installed and is using only Windows.

I followed the below commands:

``conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/develop/environment.yml python=3.9
conda activate weis-env                          # (if this does not work, try source activate weis-env)

git clone https://github.com/WISDEM/WEIS.git
 cd WEIS
 git checkout "develop"                         # (Only if you want to switch branches, say "develop")
 python setup.py develop                          # (The common "pip install -e ." will not work here)

gbarter, 2022-10-04:
You posted the commands you are using for the install, but not the output on the terminal.  Regardless, the install is likely failing because you are using regular Windows.  We have some ideas to make Windows better supported in the future using conda, but that has not been implemented yet.  For now, I would suggest using the Windows Ubuntu Subsystem Linux or installing a virtual machine.

kpramng, 2022-10-04:
**conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/develop/environment.yml python=3.9**

[image]

**conda activate weis-env**

[image]

**git clone https://github.com/WISDEM/WEIS.git**

[image]

**git checkout branch_name**  

[image]

**python setup.py develop**

(weis-env) C:\Users\prakakar\source\WEIS_Develop_Rev2\WEIS>python setup.py develop
setup.py:7: DeprecationWarning:

  `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
  of the deprecation of `distutils` itself. It will be removed for
  Python >= 3.12. For older Python versions it will remain present.
  It is recommended to use `setuptools < 60.0` for those Python versions.
  For more details, see:
    https://numpy.org/devdocs/reference/distutils_status_migration.html

  from numpy.distutils.command.build_ext import build_ext
['setup.py', 'develop']
running develop
C:\Users\prakakar\Anaconda3\envs\weis-env\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\prakakar\Anaconda3\envs\weis-env\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running build_scripts
running egg_info
running build_src
INFO: build_src
INFO: building extension "wisdem.ccblade._bem" sources
INFO: f2py options: []
INFO: f2py:> build\src.win-amd64-3.8\wisdem\ccblade\_bemmodule.c
creating build
creating build\src.win-amd64-3.8
creating build\src.win-amd64-3.8\wisdem
creating build\src.win-amd64-3.8\wisdem\ccblade
Reading fortran codes...
        Reading file 'wisdem\\ccblade\\src\\bem.f90' (format:free)
Post-processing...
        Block: _bem
                        Block: inductionfactors
In: :_bem:wisdem\ccblade\src\bem.f90:inductionfactors
get_parameters: got "eval() arg 1 must be a string, bytes or code object" on 8
                        Block: relativewind
In: :_bem:wisdem\ccblade\src\bem.f90:relativewind
get_parameters: got "eval() arg 1 must be a string, bytes or code object" on 8
                        Block: definecurvature
In: :_bem:wisdem\ccblade\src\bem.f90:definecurvature
get_parameters: got "eval() arg 1 must be a string, bytes or code object" on 8
                        Block: windcomponents
In: :_bem:wisdem\ccblade\src\bem.f90:windcomponents
get_parameters: got "eval() arg 1 must be a string, bytes or code object" on 8
                        Block: thrusttorque
In: :_bem:wisdem\ccblade\src\bem.f90:thrusttorque
get_parameters: got "eval() arg 1 must be a string, bytes or code object" on 8
                        Block: inductionfactors_dv
In: :_bem:wisdem\ccblade\src\bem.f90:inductionfactors_dv
get_parameters: got "eval() arg 1 must be a string, bytes or code object" on 8
                        Block: relativewind_dv
In: :_bem:wisdem\ccblade\src\bem.f90:relativewind_dv
[... 812 lines cut ...]
  Flags       : none
  Extra checks: none

CPU dispatch  :
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> C:\Users\prakakar\source\WEIS_Develop_Rev2\WEIS\build\temp.win-amd64-3.8\Release\ccompiler_opt_cache_ext.py
INFO:
########### EXT COMPILER OPTIMIZATION ###########
INFO: Platform      :
  Architecture: x64
  Compiler    : msvc

CPU baseline  :
  Requested   : 'min'
  Enabled     : SSE SSE2 SSE3
  Flags       : none
  Extra checks: none

CPU dispatch  :
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> C:\Users\prakakar\source\WEIS_Develop_Rev2\WEIS\WISDEM\build\temp.win-amd64-3.8\Release\ccompiler_opt_cache_ext.py

(weis-env) C:\Users\prakakar\source\WEIS_Develop_Rev2\WEIS>

gbarter, 2022-10-04:
The terminal doesn't clearly say the word, "error", which is both surprising and annoying, but the install is not working.  You can use standard Windows, sorry.

kpramng, 2022-10-05:
Hi @gbarter  Thanks a lot for getting back. I tried the installation all over again and its still the same error.

(weis-env) C:\Users\prakakar\source\WEIS_Develop_Rev2\WEIS\examples\02_control_opt>python weis_driver.py
Traceback (most recent call last):
  File "weis_driver.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
ModuleNotFoundError: No module named 'weis'

(weis-env) C:\Users\prakakar\source\WEIS_Develop_Rev2\WEIS\examples\02_control_opt>

kpramng, 2022-10-06:
Hi @gbarter I read the new Readme, [https://github.com/WISDEM/WEIS/tree/develop](url) in and saw the following updated comment `conda config --set ssl_verify no`

Can you help me when I should use this command during installation

kpramng, 2022-10-06:
Hi Garrett,

Thanks for the response. If I understand correctly, I have to install Windows Ubuntu Subsystem Linux in a virtual machine?

Best regards,
Karthik

gbarter, 2022-10-06:
You would have to use either the Windows Ubuntu Subsystem Linux **or** a virtual machine (not both).  VMWare Workstation Player would be one virtual machine option.

kpramng, 2022-10-06:
Got it. So, I can use a VMWare virtual machine and install Linux (or Ubuntu) in the Virtual
 machine?

gbarter, 2022-10-06:
Yes

kpramng, 2022-10-06:
Is there another way of automatically writing the inputs for the simulation for various DLC's (eg .fst files, .dat files etc) and obtaining the inputs other than WEIS.

gbarter, 2022-10-06:
WEIS is our preferred tool for this.  The OpenFAST team has some toolboxes to make batch editing easier that you can explore:
https://github.com/OpenFAST