NLRWindSystems/WEIS #192: WEIS/examples/07_te_flaps example does not run properly

DeepFriedDerp, 2022-09-22:
## Description
Some of the WEIS/examples do not run correctly as configured. The apparent cause is a mismatch between the format of the input yaml files (in my specific case, 07_te_flaps/modeling_options.yaml) and the modeling_schema.yaml (under WEIS/weis/inputs/) and toolbox_schema.yaml (under WEIS/ROSCO/ROSCO_toolbox/inputs/). I have not looked into other examples, so I can't speak to the extent of condition on this issue. 

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. under 07_te_flaps example directory, edit modeling_options.yaml xfoil path to point to the xfoil binary/executable
2. run dac_driver.py

### Current behavior

> Traceback (most recent call last):
  File "dac_driver.py", line 12, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
  File "/home/[userName]/gitRepos/WEIS/weis/glue_code/runWEIS.py", line 20, in run_weis
    wt_initial = WindTurbineOntologyPythonWEIS(fname_wt_input, fname_modeling_options, fname_opt_options)
  File "/home/[userName]/gitRepos/WEIS/weis/glue_code/gc_LoadInputs.py", line 26, in __init__
    self.set_weis_data()
  File "/home/[userName]/gitRepos/WEIS/weis/glue_code/gc_LoadInputs.py", line 134, in set_weis_data
    raise Exception("A distributed aerodynamic control device is defined in the geometry yaml, but the path to XFoil in the modeling options is not defined correctly")
Exception: A distributed aerodynamic control device is defined in the geometry yaml, but the path to XFoil in the modeling options is not defined correctly

- Edited python script (WEIS/weis/glue_code/gc_LoadInputs.py) to print out contents of "self.modeling_options['Level3']["xfoil"]["path"]", and it is empty, printed out results of bool(self.modeling_options['Level3']["xfoil"]["path"]), and returned false, indicating that entry is indeed empty. 
- checked modeling_schema.yaml, and xfoil parameters are not separate from Level3, as they are in the example, but listed underneath Level3 parameters.
- Rearranged the example yaml to reflect the correct location for the xfoil parameters, and the error goes away the next time I call dac_driver.py
- Am still getting errors related to ROSCO, because according to the schema yaml, the ROSCO params that are currently in the example modeling yaml have been moved to a toolbox_schema.yaml, and the example modeling yaml should only have one parameter that points to the toolbox yaml, I think. 

### Expected behavior
example runs without errors, or with errors that are dependent on user-specific paths and whatnot. 

### Code versions
- Latest commit of main branch of repo
- Python 3.8

dzalkind, 2022-09-22:
Hey Gerrit, I suspect we fixed this in #181.  It looks like the modeling options are in the correct format in the `develop` branch.  

You should be able to specify `ROSCO` modeling options in WEIS and it will check against the toolbox_schema.yaml. Feel free to post your error here.  I would try this on the `develop` branch, too.

DeepFriedDerp, 2022-09-23:
Hello, thank you for the response. 

I switched to the develop branch, re-ran the compile process via the develop branch readme, and the above issue does appear to be resolved, but new issues are popping up. 

The last error that I got for this example:

> Exception: If Flp_Mode > 0, you must set flp_kp_norm, flp_tau in the modeling options

dzalkind, 2022-09-23:
Try the settings provided in #193. I'm not 100% sure what the best parameters for that are, but maybe @nikhar-abbas or @ptrbortolotti could help us there.

nikhar-abbas, 2022-09-24:
Hi there Gerrit,

The PI controller gains are based on the following structure:
$$K_p = \frac{\alpha}{\kappa_f}$$
$$K_i = \frac{\alpha}{\kappa_f\tau}$$.

The $\alpha$ and $\tau$ terms in this formulation are the values that the user (or WEIS) needs to choose. $\kappa$ is an "efficacy" term that essentially maps the influence of a change in flap angle to a change in blade root bending moment. 

Generally speaking, the controller is not very sensitive to changes in $\tau$, so if you don't want to optimize this you can use something like 10s, as is used in the BAR [example in ROSCO](https://github.com/NREL/ROSCO/blob/51629665d22f92770f861daeebcebd47ab32abb1/Tune_Cases/BAR.yaml#L47). The $\alpha$ term does need to be tuned, however. Again, you could just start with the BAR value and modify accordingly, depending on how "optimal" you want your value to be. 

Since you seem to be using WEIS, however, you can run an optimization using DLC 1.3 and make sure that you have the inputs provided [here](https://github.com/dzalkind/WEIS/blob/8feb55001382b15d3b85abcb482ea07f383cde46/examples/07_te_flaps/modeling_options.yaml#L87) and enabling the flap related design variables in your analysis_options file with the format shown in the [schema](https://github.com/dzalkind/WEIS/blob/8feb55001382b15d3b85abcb482ea07f383cde46/weis/inputs/analysis_schema.yaml#L187). When I last used this a few months ago, this functionality worked to optimize flap controller gains. I know that the xfoil-related stuff slips through the cracks in the WEIS automated testing though, so @dzalkind or @ptrbortolotti might need to verify that they are still working if you are still getting errors.

DeepFriedDerp, 2022-09-26:
> Try the settings provided in #193. I'm not 100% sure what the best parameters for that are, but maybe @nikhar-abbas or @ptrbortolotti could help us there.

Hello,

I tried the above fix by adding the flp_kp_norm and flp_tau to the modeling options as specified in https://github.com/WISDEM/WEIS/pull/193, and got the following new error:

> Exception: 'wisdem.wt.rotorse.rs.bjs' <class BladeJointSizing>: Error calling compute(), Please add a material named joint_insert to the geometry yaml

ptrbortolotti, 2022-09-26:
Hi, thank you for the patience while we get this to work. This PR https://github.com/WISDEM/WEIS/pull/194 should fix example 07. Please give it another try
Regards,
Pietro

DeepFriedDerp, 2022-09-26:
Hi Pietro, 

I switched to https://github.com/WISDEM/WEIS/pull/194, recompiled, and attempted to run the example again. I got the following error:

> FileNotFoundError: 'xf' <class RunXFOIL>: Error calling compute(), Profile20_-10.0_Airfoil.txt not found.

ptrbortolotti, 2022-09-26:
That seems like an XFoil error... Can you please make sure that XFoil runs successfully?

DeepFriedDerp, 2022-10-10:
Hi @ptrbortolotti 

I'm not very familiar with using/installing xfoil, so I'm not very sure how to check if my particular install is working. 

I have verified it exists on my Ubuntu machine (xfoil6.97 for UNIX and win32) from the following website: https://web.mit.edu/drela/Public/web/xfoil/

I have also attempted to compile and run WEIS using a Mac, using the suggested xfoil4mac package linked on the above website. I have tried pointing the modeling yaml to both /Applications/Xfoil , /Application/Xfoil.app , and /Applications/Xfoil.app/Contents/MacOS/Xfoil , with all attempts returning some complex block of error messages. I will provide the full block of errors below. I also attempted to build xfoil6.99 from the source and instructions provided from the linked xfoil website, but was unable to properly make the final step, with an apparent error in the source code, which I'm assuming is probably an error with compiling steps on my part. 

Anyway here is the full error message from the macOS attempt, with the xfoil path in modeling_options.yaml defined as `path: /Applications/Xfoil.app/Contents/MacOS/Xfoil`

> sh: /Applications/Xfoil.app/Contents/MacOS/Xfoil: Bad CPU type in executable
Traceback (most recent call last):
  File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 2275, in _call_user_function
    yield
  File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 285, in _compute_wrapper
    self.compute(*args)
  File "/Users/agmotes/gitRepos/WEIS/weis/control/dac.py", line 357, in compute
    af_flap.af_flap_coords(self.xfoil_path, fa,  inputs['chord_start'][k],0.5,200, **xfoil_kw) #bem: the last number is the number of points in the profile.  It is currently being hard coded at 200 but should be changed to make sure it is the same number of points as the other profiles
  File "/Users/agmotes/gitRepos/WEIS/WISDEM/wisdem/ccblade/ccblade.py", line 441, in af_flap_coords
    flap_coords = np.loadtxt(saveFlnmAF)
  File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/numpy/lib/npyio.py", line 1313, in loadtxt
    arr = _read(fname, dtype=dtype, comment=comment, delimiter=delimiter,
  File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/numpy/lib/npyio.py", line 955, in _read
    fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
  File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/numpy/lib/_datasource.py", line 193, in open
    return ds.open(path, mode, encoding=encoding, newline=newline)
  File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/numpy/lib/_datasource.py", line 533, in open
    raise FileNotFoundError(f"{path} not found.")
FileNotFoundError: Profile20_-10.0_Airfoil.txt not found.

> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/Users/agmotes/gitRepos/WEIS/examples/07_te_flaps/dac_driver.py", line 12, in <module>
>     wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
>   File "/Users/agmotes/gitRepos/WEIS/weis/glue_code/runWEIS.py", line 195, in run_weis
>     wt_opt.run_model()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/problem.py", line 811, in run_model
>     self.model.run_solve_nonlinear()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 3994, in run_solve_nonlinear
>     self._solve_nonlinear()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/group.py", line 2560, in _solve_nonlinear
>     self._nonlinear_solver._solve_with_cache_check()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 26, in _solve_with_cache_check
>     self.solve()  # don't use caching
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 45, in solve
>     self._gs_iter()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/solver.py", line 787, in _gs_iter
>     subsys._solve_nonlinear()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 311, in _solve_nonlinear
>     self._compute_wrapper()
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 269, in _compute_wrapper
>     with self._call_user_function('compute'):
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/contextlib.py", line 153, in __exit__
>     self.gen.throw(typ, value, traceback)
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 2281, in _call_user_function
>     raise err_type(
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 2275, in _call_user_function
>     yield
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 285, in _compute_wrapper
>     self.compute(*args)
>   File "/Users/agmotes/gitRepos/WEIS/weis/control/dac.py", line 357, in compute
>     af_flap.af_flap_coords(self.xfoil_path, fa,  inputs['chord_start'][k],0.5,200, **xfoil_kw) #bem: the last number is the number of points in the profile.  It is currently being hard coded at 200 but should be changed to make sure it is the same nu [... cut ...]
>     fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/numpy/lib/_datasource.py", line 193, in open
>     return ds.open(path, mode, encoding=encoding, newline=newline)
>   File "/opt/anaconda3/envs/weis-env/lib/python3.10/site-packages/numpy/lib/_datasource.py", line 533, in open
>     raise FileNotFoundError(f"{path} not found.")
> FileNotFoundError: 'xf' <class RunXFOIL>: Error calling compute(), Profile20_-10.0_Airfoil.txt not found.

ptrbortolotti, 2022-10-10:
I'd recommend checking that WEIS and XFoil run indipendently first. Please try running any other WEIS example and see if it works. As for Xfoil, you could try calling it as a standalone. If it doesn't run, try following the instructions attached.
[install_xfoil.txt](https://github.com/WISDEM/WEIS/files/9748197/install_xfoil.txt)

DeepFriedDerp, 2022-10-10:
Hi @ptrbortolotti 

I rebuilt my installation of xfoil on Ubuntu in WSL, and successfully ran example 7, so it appears confirmed that these latest errors are an xfoil issue and not a WEIS issue. 

I see there are some gfortran compiler flags that are different between the public install documents and the version you sent me, I will re-attempt to build xfoil on my mac later this afternoon and document the results here for anybody running into these same issues.

DeepFriedDerp, 2022-10-12:
@ptrbortolotti 

from your install_xfoil.txt file, I got down to the command
`make -f Makefile_gfortran xfoil`

and got this error on entering said command:

> gfortran -c -O  -fdefault-real-8  ../src/xfoil.f
> gfortran -c -O  -fdefault-real-8  ../src/xpanel.f
> gfortran -c -O  -fdefault-real-8  ../src/xoper.f
> gfortran -c -O  -fdefault-real-8  ../src/xtcam.f
> gfortran -c -O  -fdefault-real-8  ../src/xgdes.f
> ../src/xgdes.f:554:20:
> 
>   540 |        CALL READR(1,ATOL,ERROR)
>       |                    2
> ......
>   554 |        CALL READR(2,XRF,ERROR)
>       |                    1
> Error: Rank mismatch between actual argument at (1) and actual argument at (2) (scalar and rank-1)
> make: *** [xgdes.o] Error 1

This is the exact same error I get when attempting to build xfoil, no matter what directions I use. It appears to be tied to the compiler identifying a source code error.

gbarter, 2022-10-12:
Hi Gerrit,

I put those install instructions together at some point to document an internal configuration.  We cannot bundle xfoil with our models because of a licensing clash. 

It is possible that your compiler needs some additional flags or that the source code is just using older conventions that your compiler no longer supports.  Issues with the xfoil source code are probably best directed at Prof Mark Drela at MIT.
https://web.mit.edu/drela/Public/web/xfoil/

Cheers,
Garrett

deslaughter, 2022-10-12:
@DeepFriedDerp, the problem is caused by newer versions of GFortran identifying an issue with the source code and treating it as an error, whereas it was previously just a warning. The previous behavior can be restored with the flag `-fallow-argument-mismatch`. This flag can be added to line 121 of `bin/Makefile_gfortran` like `DBL = -fdefault-real-8 -fallow-argument-mismatch`.

gbarter, 2022-10-12:
You rock, Derek!  Thank you so much for that tip!

DeepFriedDerp, 2022-10-12:
@ptrbortolotti @gbarter @deslaughter, that worked! I was able to install and test example 07 in WEIS successfully. I'll make sure to write all of this down for my coworkers, thanks to everyone involved!

closed 2022-10-12 by ptrbortolotti