NLRWindSystems/WEIS #198: Some examples do not run properly (getting error related to class FASTLoadCases)

Kang-SungKu, 2022-10-21:
## Description
Hello, I am interested in using WEIS as a tool to generate and evaluate floating offshore turbine models/controllers based on design parameters (e.g., design parameters for blade shape, parameters for IPC controller), such that I can test design exploration/optimization methods.

As a starting point, I am currently playing with the examples included with WEIS, but some examples do not run properly with the error related to class _FASTLoadCases_. The examples mostly fail in the functions _fast_init_ and _initialize_routines_ where input files are organized, so I believe the main reasons from mismatching between the input formats. I wonder if something is missing in the official installation procedure, or if I need to use a different branch to resolve the issue. It would be also helpful to know what is the best environment (WSL, Linux, or macOS) in terms of compatibility.

### Steps to reproduce issue
The list of the examples I found the issue. I modified _modeling_option_ to have _use_exe: False_ to find where I get the errors, but no additional modification has been made.
1. 06_IEA-15-240-RWT/weis_driver_tower_DVs.py
2. 09_design_of_experiments/doe_openfast.py
3. 11_use_bem/weis_driver.py
4. 12_linearization/doe_driver.py
5. 14_level2ccd/weis_driver.py

### Current behavior
The followings are the errors I got from each example:
1. 06_IEA-15-240-RWT/weis_driver_tower_DVs.py
```
Traceback (most recent call last):
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/drivers/doe_driver.py", line 223, in _run_case
    self._problem().model.run_solve_nonlinear()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 3994, in run_solve_nonlinear
    self._solve_nonlinear()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/group.py", line 2560, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/solver.py", line 787, in _gs_iter
    subsys._solve_nonlinear()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 311, in _solve_nonlinear
    self._compute_wrapper()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 269, in _compute_wrapper
    with self._call_user_function('compute'):
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/contextlib.py", line 153, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 2281, in _call_user_function
    raise err_type(
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 2275, in _call_user_function
    yield
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 285, in _compute_wrapper
    self.compute(*args)
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/openmdao_openfast.py", line 605, in compute
    summary_stats, extreme_table, DELs, Damage, case_list, case_name, chan_time, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/openmdao_openfast.py", line 2051, in run_FAST
    summary_stats, extreme_table, DELs, Damage, chan_time = fastBatch.run_serial()
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/runFAST_pywrapper.py", line 361, in run_serial
    _name, _ss, _et, _dl, _dam, _ct = evaluate(c)
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/runFAST_pywrapper.py", line 480, in evaluate
    return fast.execute()
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/runFAST_pywrapper.py", line 184, in execute
    openfastlib.fast_run()
  File "/mnt/c/CoDesignTurbineBlade/OpenFAST/glue-codes/python/openfast_library.py", line 226, in fast_run
    self.fast_init()
  File "/mnt/c/CoDesignTurbineBlade/OpenFAST/glue-codes/python/openfast_library.py", line 140, in fast_init
    raise RuntimeError(f"Error {_error_status.value}: {_error_message.value}")
RuntimeError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), Error 4: b'Error allocating turbine data.'
```

2. 09_design_of_experiments/doe_openfast.py
```
Traceback (most recent call last):
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/drivers/doe_driver.py", line 223, in _run_case
    self._problem().model.run_solve_nonlinear()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 3994, in run_solve_nonlinear
    self._solve_nonlinear()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/group.py", line 2560, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
[... 165 lines cut ...]
    summary_stats, extreme_table, DELs, Damage, chan_time = fastBatch.run_serial()
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/runFAST_pywrapper.py", line 361, in run_serial
    _name, _ss, _et, _dl, _dam, _ct = evaluate(c)
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/runFAST_pywrapper.py", line 480, in evaluate
    return fast.execute()
  File "/mnt/c/CoDesignTurbineBlade/weis/aeroelasticse/runFAST_pywrapper.py", line 183, in execute
    openfastlib = FastLibAPI(self.FAST_lib, os.path.abspath(os.path.basename(writer.FAST_InputFileOut)))
  File "/mnt/c/CoDesignTurbineBlade/OpenFAST/glue-codes/python/openfast_library.py", line 29, in __init__
    self._initialize_routines()
  File "/mnt/c/CoDesignTurbineBlade/OpenFAST/glue-codes/python/openfast_library.py", line 58, in _initialize_routines
    self.FAST_AllocateTurbines.argtypes = [
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/ctypes/__init__.py", line 387, in __getattr__
    func = self.__getitem__(name)
  File "/home/penelroid/anaconda3/envs/weis-env/lib/python3.10/ctypes/__init__.py", line 392, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), python: undefined symbol: FAST_AllocateTurbines
```

### Expected behavior
Should run without error

### Code versions
- The latest version from _main_ branch (as of 10/18/2022), following the official install procedure
- Tested with both Python 3.9 and 3.10
- Tested with WSL (Ubuntu 22.04.1 LTS) as well as macOS 10.13.6

dzalkind, 2022-10-21:
Hi, it's great to see you trying all these examples!   Unfortunately, some are a little more reliable than others.

The list of examples that we test in CI is [here](https://github.com/WISDEM/WEIS/blob/d6ac26e2d130a2d244a9702036ef696cb914ee1c/weis/test/test_examples_skinny.py#L4) and [here](https://github.com/WISDEM/WEIS/blob/d6ac26e2d130a2d244a9702036ef696cb914ee1c/weis/test/run_examples.py#L4).  I would probably start there for the most reliable set of examples.

All of these examples seem to be failing when OpenFAST is called.  So I wonder if OpenFAST was compiled correctly. Do you have the following files?
- <your_main_WEIS_dir>/local/lib/libopenfastlib.dylib
- <your_main_WEIS_dir>/local/bin/openfast

I generally run with `use_exe: True` to use the openfast executable (the second one) because the executable is a little more reliable in some environments.

If you don't have those files, we'll need to look at the output of your `python setup.py develop` installation to see what went wrong.

Dan

Kang-SungKu, 2022-10-25:
I appreciate your response. I confirmed that I have both _libopenfastlib.dylib_ and _openfast_, and when I use the option `use_exe: True`, I get mostly the same error ends with `Exception: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), OpenFAST Failed! Please check the run logs.`

I will focus on the most reliable examples you shared, and my first goal is to find a baseline example of a floating offshore turbine model/controller, which is standalone if possible (i.e., the entire model/controller is fully defined with the input YAML files without initial tunning), such that I can test design exploration by slightly changing some design parameters & controller parameters in the input YAML files. Please let me know if you have any suggestions, and it would be really helpful!

Best,
SungKu

closed 2023-12-22 by dzalkind