NLRWindSystems/WEIS #201: 06_IEA-15-240-RWT error

SeongwooCheon, 2022-12-04:
Hi

when I run the 06 example, the error is occurred like this:

> Using weis.aeroelasticse in ROSCO_toolbox...
> Traceback (most recent call last):
>   File "/home/default/WEIS/examples/06_IEA-15-240-RWT/weis_driver.py", line 2, in <module>
>     from weis.glue_code.runWEIS     import run_weis
>   File "/home/default/WEIS/weis/glue_code/runWEIS.py", line 7, in <module>
>     from weis.glue_code.glue_code         import WindPark
>   File "/home/default/WEIS/weis/glue_code/glue_code.py", line 22, in <module>
>     from weis.frequency.raft_wrapper import RAFT_WEIS
>   File "/home/default/WEIS/weis/frequency/raft_wrapper.py", line 6, in <module>
>     from raft.omdao_raft import RAFT_OMDAO
>   File "/home/default/WEIS/RAFT/raft/__init__.py", line 3, in <module>
>     import raft.raft_model as model
>   File "/home/default/WEIS/RAFT/raft/raft_model.py", line 11, in <module>
>     import raft.raft_fowt  as fowt
>   File "/home/default/WEIS/RAFT/raft/raft_fowt.py", line 8, in <module>
>     import pyhams.pyhams     as ph
> ModuleNotFoundError: No module named 'pyhams'
>

dzalkind, 2022-12-05:
It looks like pyhams wasn't installed properly.  Were there any errors when you ran `python setup.py develop` related to HAMS/pyhams?

SeongwooCheon, 2022-12-07:
yes, Is there a way to install HAMS/pyhams separately?

dzalkind, 2022-12-07:
The last error message looks like lapack isn't being installed.

It should have been installed here: 
```
sudo apt install gcc g++ gfortran libblas-dev liblapack-dev  -y      # (WSL only, assuming Ubuntu)
```

closed 2023-12-22 by dzalkind