NLRWindSystems/WEIS #202: Error due to removed feature of numpy (numpy.warnings)

yonghoonlee, 2023-01-14:
## Description
The execution of WEIS is not possible with new install.

With the most recent installation instruction of WEIS, the following versions are installed:
python=3.9.15
numpy=1.24.1

It seems that numpy 1.24.1 does not have numpy.warnings any more.
I checked with a bit older version of numpy (1.21.1), and it did not create any problem, since older versions have numpy.warnings.

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Install WEIS by following the installation instruction on the README.md file (from develop branch)
2. Run any example. I ran example 06.

### Current behavior
```
~/work/WEIS/examples/06_IEA-15-240-RWT develop*
weis-env ❯ python weis_driver.py     
Traceback (most recent call last):
  File "/Users/yhlee/work/WEIS/examples/06_IEA-15-240-RWT/weis_driver.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
  File "/Users/yhlee/work/WEIS/weis/glue_code/runWEIS.py", line 4, in <module>
    from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
  File "/Users/yhlee/work/WEIS/weis/glue_code/gc_LoadInputs.py", line 5, in <module>
    import weis.inputs as sch
  File "/Users/yhlee/work/WEIS/weis/inputs/__init__.py", line 1, in <module>
    from weis.inputs.validation import *
  File "/Users/yhlee/work/WEIS/weis/inputs/validation.py", line 3, in <module>
    import wisdem.inputs
  File "/Users/yhlee/work/WEIS/WISDEM/wisdem/__init__.py", line 1, in <module>
    from wisdem.glue_code.runWISDEM import run_wisdem
  File "/Users/yhlee/work/WEIS/WISDEM/wisdem/glue_code/runWISDEM.py", line 16, in <module>
    np.warnings.filterwarnings("ignore", category=np.VisibleDeprecationWarning)
  File "/Users/yhlee/opt/anaconda3/envs/weis-env/lib/python3.9/site-packages/numpy/__init__.py", line 284, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'warnings'
```

### Expected behavior

### Code versions
_List versions only if relevant_
- Python 3.9.15 (anaconda)
- Conda 22.11.1
- Numpy 1.24.1

yonghoonlee, 2023-01-14:
I am just temporarily running WEIS by installing older numpy (numpy=1.23 works). However, if numpy.warnings is deprecated, the WEIS code needs to be updated.

gbarter, 2023-01-14:
I can help with this as the same NumPy release required a number of changes in WISDEM as well.

gbarter, 2023-01-16:
I think I have updated all of the codes that would fail with the changes in Numpy v1.24 in PR #203.

closed 2023-02-01 by gbarter