NLRWindSystems/WEIS #208: Runtime errors in Main branch

DeepFriedDerp, 2023-02-02:
## Description
Errors related to Numpy occur when attempting to run example 07_te_flaps within WEIS on the main branch.

### Steps to reproduce issue

1. Install main branch of weis per the readme instructions
2. edit WEIS/examples/07_te_flaps/modeling_options.yaml to point to local xfoil executable
3. call `python dac_driver.py` while in the WEIS/examples/07_te_flaps directory

### Current behavior
> agmotes@x86_64-apple-darwin13 07_te_flaps % python dac_driver.py 
> Traceback (most recent call last):
>   File "/Users/agmotes/gitRepos/WEIS-main/examples/07_te_flaps/dac_driver.py", line 1, in <module>
>     from weis.glue_code.runWEIS     import run_weis
>   File "/Users/agmotes/gitRepos/WEIS-main/weis/glue_code/runWEIS.py", line 4, in <module>
>     from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
>   File "/Users/agmotes/gitRepos/WEIS-main/weis/glue_code/gc_LoadInputs.py", line 5, in <module>
>     import weis.inputs as sch
>   File "/Users/agmotes/gitRepos/WEIS-main/weis/inputs/__init__.py", line 1, in <module>
>     from weis.inputs.validation import *
>   File "/Users/agmotes/gitRepos/WEIS-main/weis/inputs/validation.py", line 3, in <module>
>     import wisdem.inputs
>   File "/Users/agmotes/gitRepos/WEIS-main/WISDEM/wisdem/__init__.py", line 1, in <module>
>     from wisdem.glue_code.runWISDEM import run_wisdem
>   File "/Users/agmotes/gitRepos/WEIS-main/WISDEM/wisdem/glue_code/runWISDEM.py", line 16, in <module>
>     np.warnings.filterwarnings("ignore", category=np.VisibleDeprecationWarning)
>   File "/Users/agmotes/opt/anaconda3/envs/weis-env-main/lib/python3.10/site-packages/numpy/__init__.py", line 284, in __getattr__
>     raise AttributeError("module {!r} has no attribute "
> AttributeError: module 'numpy' has no attribute 'warnings'. Did you mean: 'hanning'?

### Expected behavior
Successful run with data

DeepFriedDerp, 2023-02-09:
It turns out that numpy has a recent release (v1.24.1) that returns a run-time error when calling anything with "numpy.warnings".

Per [this thread](https://stackoverflow.com/a/74938548), the quickest way to fix this issue is to roll back to the previous version of numpy (v1.23.1). I think @gbarter recently merged a hotfix for this into the develop branch. 

I've created a pull request https://github.com/WISDEM/WEIS/pull/211 which addresses this issue, as well as the issue (https://github.com/WISDEM/WEIS/issues/205) with mat4py not installing correctly during setup.py. 

This pull request also addresses an incorrect reference in the main branch's installation instructions, which setup the weis-env according to the develop branch's environment.yaml file. Since there is significant time and number of commits between main and develop, this sets up a lot of potential for updates to develop to adversely affect the installation or execution of the main branch code.

dzalkind, 2023-02-13:
This has been addressed in #211.  Long-term support for all numpy versions is available on the `develop` branch.

closed 2023-02-13 by dzalkind