NLRWindSystems/WEIS #217: ModuleNotFoundError: No module named 'wisdem.ccblade._bem'

SYZ622, 2023-03-06:
## Description
_ModuleNotFoundError: No module named 'wisdem.ccblade._bem'_

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. I run the "WEIS\examples\03_NREL5MW_OC3_spar\weis_driver.py"
2. …
3. …

### Current behavior
I just started to contact WEIS, I don't know how to solve this problem, it's not clear if the environment is not configured correctly.
I look forward to your reply, thank you very much!
### Expected behavior
…

### Code versions
_List versions only if relevant_
- Python
- …

gbarter, 2023-03-06:
This error arises when WEIS does not install correctly.  From your path, it also looks like you are on Windows, which WEIS does not currently support.  If you are able to use the Windows Subsystem Linux, and comfortable doing so, I would start with that.

SYZ622, 2023-03-07:
@gbarter Thanks for your reply, I did not notice this situation. Thanks again.

closed 2023-03-07 by SYZ622