NLRWindSystems/WEIS #219: Example 05 fails in LandBOSSE

rafmudaf, 2023-03-13:
## Description
From the latest `develop` branch (https://github.com/WISDEM/WEIS/commit/9a4c5ce1fd2c8b5f17d8784df80859d650a8624a), example 5 fails for me with this error:

```
(weis-env) >>mbp@~/Development/WEIS/examples/05_IEA-3.4-130-RWT (develop)$ python weis_driver.py 
Using weis.aeroelasticse in ROSCO_toolbox...
Traceback (most recent call last):
  File "/Users/rmudafor/Development/WEIS/examples/05_IEA-3.4-130-RWT/weis_driver.py", line 15, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
  File "/Users/rmudafor/Development/WEIS/weis/glue_code/runWEIS.py", line 151, in run_weis
    wt_opt.setup(derivatives=False)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/problem.py", line 954, in setup
    model._setup(model_comm, mode, self._metadata)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 1474, in _setup
    self._setup_procs(self.pathname, comm, mode, self._problem_meta)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/group.py", line 595, in _setup_procs
    subsys._setup_procs(subsys.pathname, sub_comm, mode, prob_meta)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/group.py", line 595, in _setup_procs
    subsys._setup_procs(subsys.pathname, sub_comm, mode, prob_meta)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/group.py", line 595, in _setup_procs
    subsys._setup_procs(subsys.pathname, sub_comm, mode, prob_meta)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/component.py", line 218, in _setup_procs
    self.setup()
  File "/Users/rmudafor/Development/WEIS/WISDEM/wisdem/landbosse/landbosse_omdao/landbosse.py", line 57, in setup
    self.setup_discrete_inputs_that_are_dataframes()
  File "/Users/rmudafor/Development/WEIS/WISDEM/wisdem/landbosse/landbosse_omdao/landbosse.py", line 223, in setup_discrete_inputs_that_are_dataframes
    default_project_data = OpenMDAODataframeCache.read_all_sheets_from_xlsx("ge15_public")
  File "/Users/rmudafor/Development/WEIS/WISDEM/wisdem/landbosse/landbosse_omdao/OpenMDAODataframeCache.py", line 84, in read_all_sheets_from_xlsx
    xlsx = pd.ExcelFile(xlsx_filename, engine="openpyxl")
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/pandas/io/excel/_base.py", line 1695, in __init__
    self._reader = self._engines[engine](self._io, storage_options=storage_options)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/pandas/io/excel/_openpyxl.py", line 557, in __init__
    super().__init__(filepath_or_buffer, storage_options=storage_options)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/pandas/io/excel/_base.py", line 545, in __init__
    self.book = self.load_workbook(self.handles.handle)
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/pandas/io/excel/_openpyxl.py", line 568, in load_workbook
    return load_workbook(
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openpyxl/reader/excel.py", line 346, in load_workbook
    reader.read()
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openpyxl/reader/excel.py", line 303, in read
    self.parser.assign_names()
  File "/Users/rmudafor/miniconda3/envs/weis-env/lib/python3.10/site-packages/openpyxl/reader/workbook.py", line 109, in assign_names
    sheet.defined_names[name] = defn
AttributeError: 'ReadOnlyWorksheet' object has no attribute 'defined_names'
```

I noticed that it was beginning in the `landbosse` module, so I disabled the BOS flag in the modeling options and this allowed the example to run.

```yaml
WISDEM:
    BOS:
        flag: False
```

### Steps to reproduce issue

1. Install WEIS from the `develop` branch at commit https://github.com/WISDEM/WEIS/commit/9a4c5ce1fd2c8b5f17d8784df80859d650a8624a
2. In `examples/05_IEA-3.4-130-RWT`, run `python weis_driver.py`
3. ... this results in the error above
4. Change the WISDEM/BOS flag to False from True
5. Rerun `python weis_driver.py` and the example runs

### Current behavior
Example 05 aborts with a failure

### Expected behavior
Example 05 will run to completion with no failure

### Code versions
- Python 3.10.9
- WEIS v1.1 at https://github.com/WISDEM/WEIS/commit/9a4c5ce1fd2c8b5f17d8784df80859d650a8624a

gbarter, 2023-03-13:
Thanks, Raf.  I think this is the same error as on [this WISDEM Issue](https://github.com/WISDEM/WISDEM/issues/421).

The solution is to downgrade `openpyxl`.

rafmudaf, 2023-03-13:
Ah thanks @gbarter!