NLRWindSystems/WEIS #220: Please teach me about WEIS environment debugging.

SYZ622, 2023-03-20:
I'm new to WEIS and have no prior knowledge of python or wsl, so it's not clear if I've finished activating the Anaconda environment (weis-env) under wsl this way.
I'm not sure if I've finished activating the Anaconda environment (weis-env) under wsl this way. 
[image]
When I run the calculation example, I also get this Error, No module. 
[image]，
Can anybody provide some information about WEIS runtime environment configuration or something like that?
Thanks,
Best regards.

dzalkind, 2023-03-20:
It looks like you are still working in a Windows environment.  Did you install WEIS in a WSL terminal? I would try [installing](https://github.com/WISDEM/WEIS#installation) and running it there.

SYZ622, 2023-03-21:
Thanks for your reply. I haven't installed WEIS in a WSL terminal yet, should I run setup.py in the following window to install WEIS?
[image]

dzalkind, 2023-03-21:
If that's a WSL terminal, then yes.

SYZ622, 2023-03-22:
@dzalkind When I run python setup.py develop according to README, I get the following error.
`Reading https://pypi.org/simple/pytest/
Download error on https://pypi.org/simple/pytest/: [Errno 101] Network is unreachable -- Some packages may not be found!
Couldn't find index page for 'pytest' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
Download error on https://pypi.org/simple/: [Errno 101] Network is unreachable -- Some packages may not be found!
No local packages or working download links found for pytest`
`Reading https://pypi.org/simple/certifi/
Download error on https://pypi.org/simple/certifi/: [Errno 101] Network is unreachable -- Some packages may not be found!
Couldn't find index page for 'certifi' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
Download error on https://pypi.org/simple/: [Errno 101] Network is unreachable -- Some packages may not be found!
No local packages or working download links found for certifi>=2017.4.17
error: Could not find suitable distribution for Requirement.parse('certifi>=2017.4.17')`
I would like to know if this error will affect the normal operation of WEIS? And how can I fix this network problem?

closed 2023-03-23 by SYZ622