NLRWindSystems/WEIS #221: Example 03_blade analysis_options_aerostruct do not run properly

ByeongJu-Kang, 2023-04-07:
Hi
When I run the analysis_options_aerostruct, the results of optimizations are not appropriate.
The design variables of blade aero shape are changed, but the structural design variables are not changed.
Although the optimization result is successful, the constraints for AEP are not satisfied.

The analysis option like this:
[analysis_options_aerostruct.txt](https://github.com/WISDEM/WEIS/files/11176079/analysis_options_aerostruct.txt)

gbarter, 2023-04-08:
You are trying to do too much in your optimization.  You have activated all of the design variables and constraints at high levels of resolution and only giving the optimizer a short number of iterations (50) to navigate this high-dimensional design with finite differencing.  I would suggest starting simply and iteratively building in complexity.

Also, you have listed design variables and constraints in your analysis options files that are not recognized by WISDEM/WEIS.

Here is the link to valid blade aerostructural design variables:
https://github.com/WISDEM/WISDEM/blob/master/wisdem/inputs/analysis_schema.yaml#L45
Here is the link to valid blade constraints:
https://github.com/WISDEM/WISDEM/blob/master/wisdem/inputs/analysis_schema.yaml#L897

ptrbortolotti, 2023-04-12:
Hello,
see here https://github.com/WISDEM/WISDEM/issues/428
Best regards,
Pietro

closed 2023-12-22 by dzalkind