NLRWindSystems/WEIS #224: 06_IEA-15-240-RWT Errors

HCKimDaniel, 2023-05-01:
Hello, all

I installed WEIS in my work station following your instructions as below (Step 1, 2, and 3 on WSL). I am using WSL. 

>step1
>conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml python=3.9
>conda activate weis-env                          # (if this does not work, try source activate weis-env)
>sudo apt update                                  # (WSL only, assuming Ubuntu)
>
>step2
>conda config --add channels conda-forge
>sudo apt install gcc g++ gfortran libblas-dev liblapack-dev  -y      # (WSL only, assuming Ubuntu)
>
>step3
>git clone https://github.com/WISDEM/WEIS.git
>cd WEIS
>python setup.py develop                          # (The common "pip install -e ." will not work here)

In example 06, If I run python weis_driver.py, it works well, but if I try python weis_driver_TMDs.py, then following errors occur. Can you help me to resolve this problem?

```
(weis-env) hkim@WSS4HZ3291:~/WEIS/examples/06_IEA-15-240-RWT$ python weis_driver_TMDs.py
Using weis.aeroelasticse in ROSCO_toolbox...
Found existing potential model: examples/01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi
   - Trying to use this instead of running PyHAMS.
 Traceback (most recent call last):
   File "/home/hkim/WEIS/WISDEM/wisdem/glue_code/gc_PoseOptimization.py", line 370, in set_driver
     from wisdem.optimization_drivers.nlopt_driver import NLoptDriver
   File "/home/hkim/WEIS/WISDEM/wisdem/optimization_drivers/nlopt_driver.py", line 12, in <module>
     from openmdao.utils.general_utils import simple_warning
 ImportError: cannot import name 'simple_warning' from 'openmdao.utils.general_utils' (/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/utils/general_utils.py)
 
 During handling of the above exception, another exception occurred:
 
 Traceback (most recent call last):
   File "/home/hkim/WEIS/examples/06_IEA-15-240-RWT/weis_driver_TMDs.py", line 15, in <module>
     wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
   File "/home/hkim/WEIS/weis/glue_code/runWEIS.py", line 129, in run_weis
     wt_opt = myopt.set_driver(wt_opt)
   File "/home/hkim/WEIS/WISDEM/wisdem/glue_code/gc_PoseOptimization.py", line 372, in set_driver
     raise ImportError(
 ImportError: You requested an optimization method from NLopt, but need to first install NLopt to use this method.
```

dzalkind, 2023-05-01:
Hi,

From the error message, it looks like NLopt was not installed.  You can verify this with `conda list | grep nlopt` and install it using `conda install nlopt`.  I hope this helps.

If possible, we'll review our environment setup and ensure it gets installed in step 1 above.

Dan

HCKimDaniel, 2023-05-01:
Thank you for your help.

I followed your instruction, but it doesn't work. 
As you commented, nlopt is not in the list, so I installed it.
```
(base) hkim@WSS4HZ3291:~$ conda list | grep nlopt
(base) hkim@WSS4HZ3291:~$ conda install nlopt
```

I installed nlopt first 

```
(base) hkim@WSS4HZ3291:~$ conda install nlopt
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/hkim/anaconda3

  added / updated specs:
    - nlopt

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    nlopt-2.7.1                |  py310h3342525_1         415 KB  conda-forge
    python_abi-3.10            |          2_cp310           4 KB  conda-forge
    ------------------------------------------------------------
                                           Total:         419 KB

The following NEW packages will be INSTALLED:

  nlopt              conda-forge/linux-64::nlopt-2.7.1-py310h3342525_1
  python_abi         conda-forge/linux-64::python_abi-3.10-2_cp310

The following packages will be SUPERSEDED by a higher-priority channel:

  ca-certificates    pkgs/main::ca-certificates-2023.01.10~ --> conda-forge::ca-certificates-2022.12.7-ha878542_0
  certifi            pkgs/main/linux-64::certifi-2022.12.7~ --> conda-forge/noarch::certifi-2022.12.7-pyhd8ed1ab_0

Proceed ([y]/n)? y

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
```

We can see nlopt in conda list

```
(weis-env) hkim@WSS4HZ3291:~$ conda list | grep nlopt
nlopt                     2.7.1           py310h3342525_3    conda-forge
```
But the results are the same.

```
(weis-env) hkim@WSS4HZ3291:~/WEIS/examples/06_IEA-15-240-RWT$ python weis_driver_TMDs.py
Using weis.aeroelasticse in ROSCO_toolbox...
Found existing potential model: examples/01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi
    - Trying to use this instead of running PyHAMS.
Traceback (most recent call last):
  File "/home/hkim/WEIS/WISDEM/wisdem/glue_code/gc_PoseOptimization.py", line 370, in set_driver
    from wisdem.optimization_drivers.nlopt_driver import NLoptDriver
  File "/home/hkim/WEIS/WISDEM/wisdem/optimization_drivers/nlopt_driver.py", line 12, in <module>
    from openmdao.utils.general_utils import simple_warning
ImportError: cannot import name 'simple_warning' from 'openmdao.utils.general_utils' (/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/utils/general_utils.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/hkim/WEIS/examples/06_IEA-15-240-RWT/weis_driver_TMDs.py", line 15, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
  File "/home/hkim/WEIS/weis/glue_code/runWEIS.py", line 129, in run_weis
    wt_opt = myopt.set_driver(wt_opt)
  File "/home/hkim/WEIS/WISDEM/wisdem/glue_code/gc_PoseOptimization.py", line 372, in set_driver
    raise ImportError(
ImportError: You requested an optimization method from NLopt, but need to first install NLopt to use this method.
```

I remove weis-env and recreated weis-env, but the same error occurs.
Can you check one more time? 

Thank you

gbarter, 2023-05-01:
I think this is an issue with an upstream update in OpenMDAO.  I believe we resolved it in the most recent version of WISDEM, which I have now incorporated into the WEIS `develop` branch.  Could you try:
```
git checkout develop
git pull
```
(from somewhere in your WEIS directory)

HCKimDaniel, 2023-05-01:
I tried it, and there is an "ModuleNotFoundError" as below. 

```
(weis-env) hkim@WSS4HZ3291:~/WEIS/examples/06_IEA-15-240-RWT$ python weis_driver_TMDs.py
Using weis.aeroelasticse in ROSCO_toolbox...
Traceback (most recent call last):
  File "/home/hkim/WEIS/examples/06_IEA-15-240-RWT/weis_driver_TMDs.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
  File "/home/hkim/WEIS/weis/glue_code/runWEIS.py", line 7, in <module>
    from weis.glue_code.glue_code         import WindPark
  File "/home/hkim/WEIS/weis/glue_code/glue_code.py", line 23, in <module>
    from weis.frequency.raft_wrapper import RAFT_WEIS
  File "/home/hkim/WEIS/weis/frequency/raft_wrapper.py", line 6, in <module>
    from raft.omdao_raft import RAFT_OMDAO
  File "/home/hkim/WEIS/RAFT/raft/__init__.py", line 3, in <module>
    import raft.raft_model as model
  File "/home/hkim/WEIS/RAFT/raft/raft_model.py", line 17, in <module>
    import raft.raft_fowt  as fowt
  File "/home/hkim/WEIS/RAFT/raft/raft_fowt.py", line 7, in <module>
    import pyhams.pyhams     as ph
  File "/home/hkim/WEIS/pyHAMS/pyhams/pyhams.py", line 7, in <module>
    import pyhams._hams as hams
ModuleNotFoundError: No module named 'pyhams._hams'
```

HCKimDaniel, 2023-05-01:
There is no specific error in python ```setup.py develop```

```
(weis-env) hkim@WSS4HZ3291:~/WEIS$ python setup.py develop
/home/hkim/WEIS/setup.py:5: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import run_setup
/home/hkim/WEIS/setup.py:7: DeprecationWarning:

  `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
  of the deprecation of `distutils` itself. It will be removed for
  Python >= 3.12. For older Python versions it will remain present.
  It is recommended to use `setuptools < 60.0` for those Python versions.
  For more details, see:
    https://numpy.org/devdocs/reference/distutils_status_migration.html

  from numpy.distutils.command.build_ext import build_ext
['setup.py', 'develop']
running develop
/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running build_scripts
running egg_info
running build_src
INFO: build_src
INFO: building extension "wisdem.ccblade._bem" sources
INFO: f2py options: []
INFO:   adding 'build/src.linux-x86_64-3.10/build/src.linux-x86_64-3.10/wisdem/ccblade/fortranobject.c' to sources.
INFO:   adding 'build/src.linux-x86_64-3.10/build/src.linux-x86_64-3.10/wisdem/ccblade' to include_dirs.
INFO:   adding 'build/src.linux-x86_64-3.10/wisdem/ccblade/_bem-f2pywrappers.f' to sources.
INFO: building extension "wisdem.pyframe3dd._pyframe3dd" sources
INFO: building extension "wisdem.rotorse._precomp" sources
INFO: f2py options: []
INFO:   adding 'build/src.linux-x86_64-3.10/build/src.linux-x86_64-3.10/wisdem/rotorse/fortranobject.c' to sources.
INFO:   adding 'build/src.linux-x86_64-3.10/build/src.linux-x86_64-3.10/wisdem/rotorse' to include_dirs.
INFO:   adding 'build/src.linux-x86_64-3.10/wisdem/rotorse/_precomp-f2pywrappers2.f90' to sources.
INFO: build_src: building npy-pkg config files
writing WISDEM.egg-info/PKG-INFO
writing dependency_links to WISDEM.egg-info/dependency_links.txt
writing entry points to WISDEM.egg-info/entry_points.txt
writing requirements to WISDEM.egg-info/requires.txt
writing top-level names to WISDEM.egg-info/top_level.txt
/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/setuptools/command/egg_info.py:624: SetuptoolsDeprecationWarning: Custom 'build_py' does not implement 'get_data_files_without_manifest'.
Please extend command classes from setuptools instead of distutils.
  warnings.warn(
reading manifest file 'WISDEM.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'WISDEM.egg-info/SOURCES.txt'
running build_ext
INFO: customize UnixCCompiler
INFO: customize UnixCCompiler using build_ext
INFO: CCompilerOpt.__init__[813] : load cache from file -> /home/hkim/WEIS/WISDEM/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py
INFO: CCompilerOpt.__init__[824] : hit the file cache
INFO: get_default_fcompiler: matching types: '['arm', 'gnu95', 'intel', 'lahey', 'pg', 'nv', 'absoft', 'nag', 'vast', 'compaq', 'intele', 'intelem', 'gnu', 'g95', 'pathf95', 'nagfor', 'fujitsu']'
INFO: customize ArmFlangCompiler
WARN: Could not locate executable armflang
INFO: customize Gnu95FCompiler
INFO: Found executable /usr/bin/gfortran
INFO: customize Gnu95FCompiler
INFO: customize Gnu95FCompiler using build_ext
Creating /home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/WISDEM.egg-link (link to .)
Adding WISDEM 3.6.1 to easy-install.pth file
Installing compare_designs script to /home/hkim/anaconda3/envs/weis-env/bin
Installing wisdem script to /home/hkim/anaconda3/envs/weis-env/bin

Installed /home/hkim/WEIS/WISDEM
Processing dependencies for WISDEM==3.6.1
Searching for pytest
Reading https://pypi.org/simple/pytest/
[... 1064 lines cut ...]
  Extra checks: none

CPU dispatch  :
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /home/hkim/WEIS/pyHAMS/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py
INFO:
########### EXT COMPILER OPTIMIZATION ###########
INFO: Platform      :
  Architecture: x64
  Compiler    : gcc

CPU baseline  :
  Requested   : 'min'
  Enabled     : SSE SSE2 SSE3
  Flags       : -msse -msse2 -msse3
  Extra checks: none

CPU dispatch  :
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /home/hkim/WEIS/WISDEM/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py
```

HCKimDaniel, 2023-05-01:
@gbarter  
There are two errors in ```python setup.py develop```

error1

```
Traceback (most recent call last):
  File "/home/hkim/WEIS/WISDEM/setup.py", line 104, in <module>
    run_meson_build(staging_dir)
  File "/home/hkim/WEIS/WISDEM/setup.py", line 44, in run_meson_build
    raise OSError("The meson command cannot be found on the system")
OSError: The meson command cannot be found on the system
```

error 2
```
Traceback (most recent call last):
  File "/home/hkim/WEIS/pyHAMS/setup.py", line 90, in <module>
    run_meson_build(staging_dir)
  File "/home/hkim/WEIS/pyHAMS/setup.py", line 38, in run_meson_build
    p1 = subprocess.run(sysargs, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "/home/hkim/anaconda3/envs/weis-env/lib/python3.10/subprocess.py", line 503, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/hkim/anaconda3/envs/weis-env/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/hkim/anaconda3/envs/weis-env/lib/python3.10/subprocess.py", line 1847, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'None'
```

Following is whole output in ```python setup.py develop``` 

```
(weis-env) hkim@WSS4HZ3291:~/WEIS$ python setup.py develop
/home/hkim/WEIS/setup.py:5: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import run_setup
/home/hkim/WEIS/setup.py:7: DeprecationWarning:

  `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
  of the deprecation of `distutils` itself. It will be removed for
  Python >= 3.12. For older Python versions it will remain present.
  It is recommended to use `setuptools < 60.0` for those Python versions.
  For more details, see:
    https://numpy.org/devdocs/reference/distutils_status_migration.html

  from numpy.distutils.command.build_ext import build_ext
['setup.py', 'develop']
Traceback (most recent call last):
  File "/home/hkim/WEIS/WISDEM/setup.py", line 104, in <module>
    run_meson_build(staging_dir)
  File "/home/hkim/WEIS/WISDEM/setup.py", line 44, in run_meson_build
    raise OSError("The meson command cannot be found on the system")
OSError: The meson command cannot be found on the system
running develop
/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing rosco.egg-info/PKG-INFO
writing dependency_links to rosco.egg-info/dependency_links.txt
writing requirements to rosco.egg-info/requires.txt
writing top-level names to rosco.egg-info/top_level.txt
reading manifest file 'rosco.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'rosco.egg-info/SOURCES.txt'
running build_ext
Creating /home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/rosco.egg-link (link to .)
rosco 2.6.0 is already the active version in easy-install.pth

[... 1143 lines cut ...]
Using /home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages
Searching for six==1.16.0
Best match: six 1.16.0
Adding six 1.16.0 to easy-install.pth file

Using /home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages
Finished processing dependencies for WEIS==1.1
INFO:
########### EXT COMPILER OPTIMIZATION ###########
INFO: Platform      :
  Architecture: x64
  Compiler    : gcc

CPU baseline  :
  Requested   : 'min'
  Enabled     : SSE SSE2 SSE3
  Flags       : -msse -msse2 -msse3
  Extra checks: none

CPU dispatch  :
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /home/hkim/WEIS/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py
```

HCKimDaniel, 2023-05-01:
@gbarter 

I removed the environment, created new environment again, and used the "develop" branch. 
In this branch, I got some errors as below in `python setup.py develop`. These errors didn't occur in previous case. 
I posted whole results from conda env create to run example for "weis_driver.py for 06 Example"

```(base) hkim@WSS4HZ3291:~$ conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: - Ran pip subprocess with arguments:
['/home/hkim/anaconda3/envs/weis-env/bin/python', '-m', 'pip', 'install', '-U', '-r', '/tmp/condaenv.9ydvt3h1.requirements.txt', '--exists-action=b']
Pip subprocess output:
Collecting fatpack (from -r /tmp/condaenv.9ydvt3h1.requirements.txt (line 1))
  Using cached fatpack-0.7.7-py3-none-any.whl (18 kB)
Collecting marmot-agents (from -r /tmp/condaenv.9ydvt3h1.requirements.txt (line 2))
  Using cached marmot_agents-0.2.5-py3-none-any.whl (39 kB)
Collecting mat4py (from -r /tmp/condaenv.9ydvt3h1.requirements.txt (line 3))
  Using cached mat4py-0.5.0-py2.py3-none-any.whl (13 kB)
Requirement already satisfied: numpy in ./lib/python3.10/site-packages (from fatpack->-r /tmp/condaenv.9ydvt3h1.requirements.txt (line 1)) (1.23.1)
Installing collected packages: mat4py, marmot-agents, fatpack
Successfully installed fatpack-0.7.7 marmot-agents-0.2.5 mat4py-0.5.0

done
#
# To activate this environment, use
#
#     $ conda activate weis-env
#
# To deactivate an active environment, use
#
#     $ conda deactivate

(base) hkim@WSS4HZ3291:~$ conda activate weis-env
(weis-env) hkim@WSS4HZ3291:~$ sudo apt update
[sudo] password for hkim:
Hit:1 http://archive.ubuntu.com/ubuntu jammy InRelease
Get:2 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]
Get:3 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [108 kB]
Get:4 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [912 kB]
Get:6 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 c-n-f Metadata [18.7 kB]
Get:7 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 c-n-f Metadata [14.3 kB]
Fetched 1282 kB in 11s (118 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
102 packages can be upgraded. Run 'apt list --upgradable' to see them.
(weis-env) hkim@WSS4HZ3291:~$ conda config --add channels conda-forge
Warning: 'conda-forge' already in 'channels' list, moving to the top
(weis-env) hkim@WSS4HZ3291:~$ sudo apt install gcc g++ gfortran libblas-dev liblapack-dev  -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
gfortran is already the newest version (4:11.2.0-1ubuntu1).
libblas-dev is already the newest version (3.10.0-2ubuntu1).
liblapack-dev is already the newest version (3.10.0-2ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 102 not upgraded.
(weis-env) hkim@WSS4HZ3291:~$ git clone https://github.com/WISDEM/WEIS.git
Cloning into 'WEIS'...
remote: Enumerating objects: 35541, done.
remote: Counting objects: 100% (2047/2047), done.
remote: Compressing objects: 100% (1240/1240), done.
[... 1600 lines cut ...]
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /home/hkim/WEIS/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py

(weis-env) hkim@WSS4HZ3291:~/WEIS$ cd examples/
(weis-env) hkim@WSS4HZ3291:~/WEIS/examples$ cd 06_IEA-15-240-RWT/
(weis-env) hkim@WSS4HZ3291:~/WEIS/examples/06_IEA-15-240-RWT$ python weis_driver.py
/home/hkim/anaconda3/envs/weis-env/lib/python3.10/site-packages/pyoptsparse/pyOpt_MPI.py:68: UserWarning: mpi4py could not be imported. mpi4py is required to use the parallel gradient analysis and parallel objective analysis for non-gradient based optimizers. Continuing using a dummy MPI module from pyOptSparse.
  warnings.warn(warn)
Traceback (most recent call last):
  File "/home/hkim/WEIS/examples/06_IEA-15-240-RWT/weis_driver.py", line 2, in <module>
    from weis.glue_code.runWEIS     import run_weis
  File "/home/hkim/WEIS/weis/glue_code/runWEIS.py", line 4, in <module>
    from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
  File "/home/hkim/WEIS/weis/glue_code/gc_LoadInputs.py", line 5, in <module>
    import weis.inputs as sch
  File "/home/hkim/WEIS/weis/inputs/__init__.py", line 1, in <module>
    from weis.inputs.validation import *
  File "/home/hkim/WEIS/weis/inputs/validation.py", line 3, in <module>
    import wisdem.inputs
ModuleNotFoundError: No module named 'wisdem'
(weis-env) hkim@WSS4HZ3291:~/WEIS/examples/06_IEA-15-240-RWT$

```

gbarter, 2023-05-02:
Sorry for the hassle in creating a new environment.  Did you create a new environment from the environment file?  It looks like `meson` wasn't installed, which is necessary for installing some of these other tools.  You can try `conda install meson`, but a more robust approach might be: `conda env update --file environment.yml`.  I just went through a did a bit of cleanup on the list of packages to make sure it was up to date, so you can do a `git pull` now that you are on the `develop` branch to get the latest updates.

HCKimDaniel, 2023-05-02:
@gbarter 

Thank you for your time and help. I tried

`conda env update --file environment.yml`

Now example of weis_driver_TMDs.py in 06_IEA-15-240-RWT works well. 

Thank you so much!!

closed 2023-05-02 by HCKimDaniel