NLRWindSystems/WEIS #226: It doesn't finish in Example 02.

HCKimDaniel, 2023-05-05:
Hello,

I am running "controller optimization only case" based on ["Control Only Optimization #147"] (https://github.com/WISDEM/WEIS/pull/147)
But it doesn't finish in running of example/02_control_opt in the status below. 

All the output files for all the cases (25 cases) are saved in rank_0 folder, but it is not processed after that like below. I included only few lines for starting point and ending point. Can you help me to finish this process? 

`(weis-env) hkim@WSS4HZ3291:~/WEIS/weis/test$ mpirun -np 32 python test_examples_skinny.py

NOW RUNNING: 02_control_opt/weis_driver

weis_driver /home/hkim/WEIS/examples/02_control_opt/weis_driver.py

NOW RUNNING: 02_control_opt/weis_driver

weis_driver /home/hkim/WEIS/examples/02_control_opt/weis_driver.py
.....

Runtime:        IEA15_07.fst = 735.22s
Generator speed:    7.5 RPM, Pitch angle:   9.2 deg, Power: 14967.0 kW, Est. wind Speed:  13.5 m/s
 Time: 710 of 720 seconds.  Estimated final completion at 08:58:52.                               Generator speed:    7.5 RPM, Pitch angle:   3.7 deg, Power: 14620.0 kW, Est. wind Speed:  11.1 m/s

  Total Real Time:       12.318 minutes
  Total CPU Time:        12.288 minutes
  Simulation CPU Time:   11.997 minutes
  Simulated Time:        12 minutes
  Time Ratio (Sim/CPU):  1.0002

  OpenFAST terminated normally.

Runtime:        IEA15_11.fst = 739.58s
Generator speed:    7.6 RPM, Pitch angle:   9.2 deg, Power: 15011.9 kW, Est. wind Speed:  13.6 m/s

  Total Real Time:       12.53 minutes
  Total CPU Time:        12.497 minutes
  Simulation CPU Time:   12.199 minutes
  Simulated Time:        12 minutes
  Time Ratio (Sim/CPU):  0.98369

  OpenFAST terminated normally.

Runtime:        IEA15_21.fst = 751.86s

`

dzalkind, 2023-05-09:
This example does not use much of WISDEM, including the `rotorse` module.  That is why the response is missing.

You will need to add the objective, [like it is here](https://github.com/WISDEM/WEIS/blob/e5e124c7e2340f4392127ba40f256486d5f97d40/WISDEM/wisdem/glue_code/gc_PoseOptimization.py#L431), on the [weis pose optimization method](https://github.com/WISDEM/WEIS/blob/e5e124c7e2340f4392127ba40f256486d5f97d40/weis/glue_code/gc_PoseOptimization.py#L72), with `aeroelastic.AEP` as the objective.  It's probable that you'll also need logic to detect which objective to actually use within openmdao, or it will try to have both.

It's also likely that using AEP as a merit figure, without constraints on loading, will give you a controller that does not work well and may be unstable. Though if you keep the constraint on the generator speed, that should keep things somewhat stable.

HCKimDaniel, 2023-05-10:
Thank you for your answer. I will try it.

closed 2023-05-11 by HCKimDaniel