NLRWindSystems/WEIS #230: How can I add multiple breakpoints in WEIS?

khc4156, 2023-06-06:
Hello, 

Thank you for your time always. 

Now I am following the paper below with WEIS. 

Zalkind, D., Abbas, N.J., Jasa, J., Wright, A. and Fleming, P., 2022, May. Floating wind turbine control optimization. In Journal of Physics: Conference Series (Vol. 2265, No. 4, p. 042021). IOP Publishing.

In the paper, there is an explanation of the multiple breakpoints to define natural frequency (omega) and damping ratio (zeta). 
I want to practice this in WEIS, but I couldn't find which input option is related to this. 

Can you let me know which input option I can add or change for using the different number of breakpoints in WEIS?

Thank you.

Daniel Kim

dzalkind, 2023-06-06:
No, sorry :-) 

This is it: https://github.com/WISDEM/WEIS/blob/e5e124c7e2340f4392127ba40f256486d5f97d40/examples/02_control_opt/modeling_options.yaml#L93

closed 2023-06-07 by khc4156

wind2021, 2023-09-06:
Dear Daniel, 
Thank you for all your help. I have two questions. 
As shown in the figure below, in the simulation results obtained by running **02_control_opt** with statements（python weis_driver.py）, I found that my iterations were small，just 1 iteration？ Should I change parameters such as **max_major_iter**、**max_minor_iter**、**max_iter** all to 100 to get the Figure 2 result？
What should I do if I want to use this method for NREL 5MW_OC4_semi turbine？
[image]
[image]

Best Regards
Lei Xue

dzalkind, 2023-09-06:
Change `max_iter` to increase the iterations.  However, it depends on the specific solver, so you may need to try them out if you change the solver.

wind2021, 2023-09-07:
Thank you！I just change the max_iter and it runs well. 
But can you give me some advice about how to use this method for **NREL 5MW_OC4_semi** turbine model？As there is only **IEA-15-240-RWT-UMaineSemi** turbine model in **example 02_control_opt**

dzalkind, 2023-09-08:
If you already have an OpenFAST model for the NREL_5MW_OC4_semi, you can assign it in the modeling options of the `02_control_opt` example.

Alternatively, the `04` example will generate an OpenFAST model from the geometry input provided in that example.  You can use the same (or similar) analysis options in that example as in the `02` example you have been using.

dzalkind, 2023-09-21:
It looks like the library for pyframe3dd is missing.  This library is generated during the installation of WISDEM.  You can navigate to the `WISDEM/` directory and run `python setup.py develop` there.  Please share any errors you see and any setup logs within the `WISDEM/build` directory.

wind2021, 2023-09-21:
[image]
[image]
[image]
[image]

As you can see this. What should I do?

wind2021, 2023-09-21:
[image]
Is this the  pyframe3dd ?

dzalkind, 2023-09-21:
If you're using a branch that used meson, which it seems like you are, there should be a `WISDEM/meson_build/setup.log` file with more information.

wind2021, 2023-09-21:
[image]
In fact, I initially installed WSL on disk C and later moved to disk D. Whether it makes a difference?

dzalkind, 2023-09-21:
Try deleting the `meson_build` directory, any previous `build` directories in `WISDEM/`, and `WISDEM/WISDEM.egg-info`, then try `python setup.py develop` again.

wind2021, 2023-09-21:
[image]
Thank you very much！Now the example 04 runs successfully, but when I put the output file **refturb_output.yaml**  to 02_control_opt and run example 02, it appeared RuntimeError: 'wisdem.wt.floatingse.load.frame' as the picture.  What should I do now?