NLRWindSystems/WEIS #231: Installation on WSL

Gawage, 2023-06-09:
I am struggling to install WEIS on Ubuntu with WSL. 

Following the instructions in the readme document I get the following error report generated when creating the environment.
_conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml python=3.9_

I am new to WSL and am unsure what this issue might be - is anyone able to help?

Error message below...

Collecting package metadata (repodata.json): failed

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/requests/adapters.py", line 456, in send
        conn = self.get_connection(request.url, proxies)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/requests/adapters.py", line 345, in get_connection
        proxy = prepend_scheme_if_needed(proxy, "http")
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/requests/utils.py", line 988, in prepend_scheme_if_needed
        parsed = parse_url(url)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/urllib3/util/url.py", line 397, in parse_url
        return six.raise_from(LocationParseError(source_url), None)
      File "<string>", line 3, in raise_from
    urllib3.exceptions.LocationParseError: Failed to parse: https://id:pw@address:port

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/exceptions.py", line 1132, in __call__
        return func(*args, **kwargs)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda_env/cli/main.py", line 78, in do_call
        exit_code = getattr(module, func_name)(args, parser)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/notices/core.py", line 121, in wrapper
        return func(*args, **kwargs)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda_env/cli/main_create.py", line 152, in execute
        result[installer_type] = installer.install(prefix, pkg_specs, args, env)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda_env/installers/conda.py", line 53, in install
        unlink_link_transaction = solver.solve_for_transaction(
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/solve.py", line 134, in solve_for_transaction
        unlink_precs, link_precs = self.solve_for_diff(update_modifier, deps_modifier,
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/solve.py", line 177, in solve_for_diff
        final_precs = self.solve_final_state(update_modifier, deps_modifier, prune, ignore_pinned,
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/solve.py", line 282, in solve_final_state
        ssc = self._collect_all_metadata(ssc)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/common/io.py", line 84, in decorated
        return f(*args, **kwds)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/solve.py", line 449, in _collect_all_metadata
        index, r = self._prepare(prepared_specs)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/solve.py", line 1064, in _prepare
        reduced_index = get_reduced_index(self.prefix, self.channels,
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/index.py", line 267, in get_reduced_index
        new_records = SubdirData.query_all(spec, channels=channels, subdirs=subdirs,
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/subdir_data.py", line 169, in query_all
        result = tuple(chain.from_iterable(executor.map(subdir_query, channel_urls)))
      File "/home/georgee/anaconda3/lib/python3.10/concurrent/futures/_base.py", line 621, in result_iterator
        yield _result_or_cancel(fs.pop())
      File "/home/georgee/anaconda3/lib/python3.10/concurrent/futures/_base.py", line 319, in _result_or_cancel
        return fut.result(timeout)
      File "/home/georgee/anaconda3/lib/python3.10/concurrent/futures/_base.py", line 458, in result
        return self.__get_result()
      File "/home/georgee/anaconda3/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
        raise self._exception
      File "/home/georgee/anaconda3/lib/python3.10/concurrent/futures/thread.py", line 58, in run
        result = self.fn(*self.args, **self.kwargs)
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/subdir_data.py", line 158, in subdir_query
        return tuple(
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/subdir_data.py", line 174, in query
        self.load()
      File "/home/georgee/anaconda3/lib/python3.10/site-packages/conda/core/subdir_data.py", line 264, in load
        _internal_state = self._load()
[... 58 lines cut ...]
       base environment : /home/georgee/anaconda3  (writable)
      conda av data dir : /home/georgee/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/georgee/anaconda3/pkgs
                          /home/georgee/.conda/pkgs
       envs directories : /home/georgee/anaconda3/envs
                          /home/georgee/.conda/envs
               platform : linux-64
             user-agent : conda/23.3.1 requests/2.28.1 CPython/3.10.9 Linux/5.15.90.1-microsoft-standard-WSL2 ubuntu/22.04.2 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.

If submitted, this report will be used by core maintainers to improve
future releases of conda.
Would you like conda to send this report to the core maintainers? [y/N]:
Timeout reached. No report sent.

gbarter, 2023-06-09:
I don't think this is a WEIS issue.  It looks like your WSL is having network connectivity issues with the outside world.  Maybe try some basic `ping` tests from the WSL terminal to check?

fronterapp, 2023-07-14:
I face a similar problem when trying to create the WEIS environment in WSL. This should have little to do with WEIS itself since this step is executed well before WEIS installation. 

The `Collecting package metadata` step runs fine, but then the `Solving Environment` step ends abruptly without raising an error:

```
(base) peref@BN-PF:~$ conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml python=3.9.12
Collecting package metadata (repodata.json): done
Solving environment: \ (base) peref@BN-PF:~$
```

Any clues on what may be going on?

gbarter, 2023-07-14:
@fronterapp - That is a great tip to add the `-nodefaults` to the yaml-file.  I will try to make that change on the backend.  Here at NREL, we are moving more and more to [mamba](https://mamba.readthedocs.io/en/latest/installation.html) to manage the `conda` commands given how often they seem to struggle.