NLRWindSystems/WEIS #232: Error when importing from attr

alexandermoennig, 2023-07-24:
Hi, I am trying to get WEIS running on Debian. However, I am struggling to resolve the following error when running example 05 (and all the others I tried so far):
```
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/alexander/Dokumente/Master/Masterarbeit/Code/WEIS/weis/glue_code/runWEIS.py", line 4, in <module>
    from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/Dokumente/Master/Masterarbeit/Code/WEIS/weis/glue_code/gc_LoadInputs.py", line 5, in <module>
    import weis.inputs as sch
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/Dokumente/Master/Masterarbeit/Code/WEIS/weis/inputs/__init__.py", line 1, in <module>
    from weis.inputs.validation import *
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/Dokumente/Master/Masterarbeit/Code/WEIS/weis/inputs/validation.py", line 2, in <module>
    import jsonmerge
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/jsonmerge/__init__.py", line 4, in <module>
    from jsonmerge.resolver import LocalRefResolver
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/jsonmerge/resolver.py", line 2, in <module>
    from jsonschema.validators import RefResolver, urldefrag, urljoin
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/jsonschema/__init__.py", line 13, in <module>
    from jsonschema._format import FormatChecker
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/jsonschema/_format.py", line 11, in <module>
    from jsonschema.exceptions import FormatError
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/jsonschema/exceptions.py", line 14, in <module>
    from attrs import define
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/alexander/anaconda3/envs/weis-envir/lib/python3.9/site-packages/attrs/__init__.py", line 3, in <module>
    from attr import (
ImportError: cannot import name 'AttrsInstance' from 'attr' (/home/alexander/.local/lib/python3.9/site-packages/attr/__init__.py)
python-BaseException
Backend QtAgg is interactive backend. Turning interactive mode on.
```

I already reinstalled the environment about three times, with no effect.
Since the given command for creating the environment didn't work for me (`SpecNotFound: Invalid name 'python=3.9', try the format: user/package`), I downloaded the environment.yml and added `python=3.9.*` to the list instead. Other than that, I just followed the given instructions. 
As far as I can tell, the installation of WEIS itself (running the setup.py) seems to work without any issue, though.
I would appreciate any help. Thank you.

dzalkind, 2023-07-24:
Hi,

You are correct about the python specification breaking the installation process.  We have made that change in the `develop` branch of WEIS.   Can you try a fresh install of the environment using the instructions on the develop branch?  If you still have trouble, please attach a listing of your conda install environment with `conda list`.  Note that the develop branch uses a new package builder, meson, so the logs will be in `meson_build` directories. 

Best, Dan

alexandermoennig, 2023-07-24:
Hi Dan,
that solved it! Thank you very much for your help.

closed 2023-07-24 by alexandermoennig