NLRWindSystems/WEIS #233: DISCON.IN output of example 05

alexandermoennig, 2023-07-27:
Hi,
I am currently working with the example 05 on the develop branch and noticed some discrepancies between the controller output (DISCON.IN) generated by WEIS and the one provided in the repository of the RWT.
Using the ROSCO toolbox and both DISCON.IN files, I generated the following plot:
[image]
My question is whether these differences are expected when using WEIS, or if there's a possibility to correct them?

Thanks in advance.
Kind regards,
Alex

dzalkind, 2023-07-27:
Hi Alex,

These min pitch tables depend on the Cp, Ct surfaces.  They are set up in WEIS to be very coarse, which usually leads to controller issues.

In the modeling options, you should set:
`n_tsr_perf_surfaces: 25`
`n_pitch_perf_surfaces: 25`

I hope this helps.

Best, Dan

alexandermoennig, 2023-07-27:
Hi Dan,

yes, that helped. Looks perfect now:
[image]
Thank you very much for your help :) 

Kind regards,
Alex

closed 2023-07-27 by alexandermoennig