NLRWindSystems/WEIS #237: InflowWind.dat file has unexpected variable

mlidGWT, 2023-10-18:
## Description
I recently updated my WEIS/develop repo to the latest commit and now a case I've run successfully before the update is no longer working. The InflowWind.dat file (written by WEIS) has the new `VelInterpCubic` parameter on line 8, but OpenFAST/WEIS seems to not be expecting it, instead looking for `NWindVel` on line 8, which is now on line 9. I imagine this issue will also happen for any of the other parameters updated at the same time as indicated [here](https://openfast.readthedocs.io/en/main/source/user/api_change.html#openfast-v3-4-0-to-openfast-v3-5-0).
It therefore seems like the WEIS/OpenFAST file writer has been updated to v3.5.0, but the file reader/interpreter is still at v3.4.0.

### Steps to reproduce issue
I was originally working in my own directory with the IEA15MW RWT, but I have confirmed that the behavior is the same when running example 06. 

1. Update WEIS/develop to latest commit using `$ git pull` within my weis-env and WEIS/ folder
2. Recompile WEIS using `$ python setup.py develop` 
3. Open a new terminal window and reactivate weis-env
4. cd to WEIS/examples/06_IEA-15-240-RWT/
5. Run example with `$ python weis_driver.py`

Please advise if any other steps are needed after updating and before running WEIS - there is no documentation around how to update WEIS. 

### Current behavior
After about a minute of WEIS doing it's thing, an error occurs: 

> FAST_InitializeAll:InflowWind_Init:InflowWind_ParseInputFileInfo:ParseInVar:
  >> A fatal error occurred when parsing data from "./IEA15_0_InflowFile.dat".
  >> The variable "NWindVel" was not found on line #8.

> OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

 > Aborting OpenFAST.

There are also two Traceback errors, which conclude with the same determination that "NWindVel" was not found on line #8 (included in attached file: [WEIS-error_NWindVel_Traceback.txt](https://github.com/WISDEM/WEIS/files/13034162/WEIS-error_NWindVel_Traceback.txt)). 

### Expected behavior
OpenFAST and WEIS execute the example as expected.

### Code versions
- WEIS/develop @ https://github.com/WISDEM/WEIS/commit/2dd692a019ee01a3cd69e5b60819b12de002d606
- Python v3.10.11

dzalkind, 2023-10-18:
Hi,

Since OpenFAST is still looking for `NWindVel` on line 8, and the writer is putting it on line 9 (consistent with v3.5.0), it seems like the OpenFAST executable was not updated.  It's possible the installation and compilation of OpenFAST was not successful.  

You could try re-installing and looking for any OpenFAST compilation errors.  Alternatively, you could `conda install openfast=3.5.0` and then add the path to OpenFAST manually to the modeling options, like the code snippet below.

We are moving towards conda installations of most tools, so this will be the default at some point in the near future.  OpenFAST will likely be the first (and most helpful) subtree we do this with.

Best, Dan

```
General:
    verbosity: False  # When set to True, the code prints to screen many infos
    openfast_configuration:
        save_timeseries: True
        save_iterations: True
        use_exe: True
        FAST_exe: /Users/dzalkind/opt/anaconda3/envs/rosco-env/bin/openfast
```

mlidGWT, 2023-10-19:
Thank you, this worked. For my own understanding, I didn't think WEIS was using the executable before specifying it with `use_exe: True`, and when I run OpenFAST directly I also don't use the executable. Does this mean my installed version of OpenFAST needs to be updated as well? In my initial issue was it writing the files with my compiled version of OpenFAST, at v3.5.0, but then trying to run it with an outdated executable?

dzalkind, 2023-10-19:
WEIS can run OpenFAST either through an executable or a dynamic library.  If there are issues, I like to use the executable as it is a little more straightforward to set up.  You should not need to update or install another version of OpenfAST.  

Your issue before, I suspect, was that version 3.5.0 of OpenFAST did not actually compile, and it was still the same v3.4.0 OpenFAST trying to read a v3.5.0 input, written by WEIS.

Best, Dan