NLRWindSystems/WEIS #238: Unable to run WEIS

Russell9798, 2023-10-19:
Hi,

I have been trying to install WEIS for the first time using WSL and followed the installation guide. I think I've successfully completed the installation but whenever I try to run any of the .py scripts, for example runWEIS.py, I get the following error:

ModuleNotFoundError: No module named 'weis'

I can't seem to figure out how to import weis as a module. Is there a way I can do this? Please forgive my ignorance if this is a simple fix, I don't have much experience with python/WSL.

Many thanks,
Andrew

Russell9798, 2023-10-19:
Hi Dan,

I suspected that may have been the case. I can't see any errors in the output when running `python setup.py develop`, only warnings. I'm working from the main branch.

After running:
`python`
`import wisdem`

I get the following error:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/c/Users/s2113132/WEIS/WISDEM/wisdem/__init__.py", line 1, in <module>
    from wisdem.glue_code.runWISDEM import run_wisdem
  File "/mnt/c/Users/s2113132/WEIS/WISDEM/wisdem/glue_code/runWISDEM.py", line 16, in <module>
    np.warnings.filterwarnings("ignore", category=np.VisibleDeprecationWarning)
  File "/usr/local/lib/python3.10/dist-packages/numpy/__init__.py", line 333, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'warnings'. Did you mean: 'hanning'?`

Russell9798, 2023-10-23:
Hi Dan,

I am trying again to install WEIS, starting with creating the Anaconda environment. When I run

`conda create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml python=3.9`

I get the following error:

"Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - //raw.githubusercontent.com/wisdem/weis/main/environment.yml

Current channels:

  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch "

How might I be able to fix this?

gbarter, 2023-10-23:
The default `conda` tool continues to suffer.  I would recommend installing the [Miniforge](https://github.com/conda-forge/miniforge#miniforge3) distribution and trying again.

Russell9798, 2023-10-23:
Hi Garrett,

I tried again with Miniforge installed but unfortunately it led to the same error. Is there anything else I could try?

gbarter, 2023-10-23:
What operating system are you on?  You might also try dropping the `python=3.9` part.

Russell9798, 2023-10-23:
I'm using Windows sub-system for Linux. It worked in building the environment afterI dropped the `python=3.9` part, thanks! This worked for me:

`conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml`

I've now followed all the other steps of the installation guide up to

`python setup.py develop`

But now I'm getting the same error again as mentioned in the first comment - when I try to run any of the .py scripts, for example, runWEIS.py, I get the following error:

ModuleNotFoundError: No module named 'weis'

Do you have any suggestions for how I can fix this?

dzalkind, 2023-10-23:
Hi Andrew,

If you're running from the `main` branch, you may need to install an earlier version of `numpy`.  E.g., `conda install numpy=1.22`.  Alternatively, you can update to the `develop` branch and try re-installing.  The develop branch should be compatible with the latest numpy.

Like before, it's best to check the individual modules of WEIS when debugging the installation.  Your message about WISDEM gave me the numpy clue.

Dan

Russell9798, 2023-10-24:
Hi Dan,

I think I've got it working now. As you suggested, I installed an earlier version of `numpy` via `conda install numpy=1.22`. 

However, this then gave me the error of: 
`RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xf`

To fix this I ran:
`pip install numpy --upgrade`

Which returned the error
`AttributeError: module 'numpy' has no attribute 'warnings'. Did you mean: 'hanning'?`

To fix this, I then ran
`pip3 install mxnet-mkl==1.6.0 numpy==1.23.1`

And now I'm able to run the examples. Thanks so much for your help!

Andrew

Russell9798, 2023-10-24:
For future reference, it might be useful to outline the steps all the steps I took for installation using WSL.

1. Install WSL using instructions from https://learn.microsoft.com/en-us/windows/wsl/install. I had to downgrade to WSL 1 using `wsl --set-version Ubuntu 1`
2. Install Windows Terminal from https://learn.microsoft.com/en-us/windows/terminal/install
3. Open WSL within Windows Terminal using `wsl.exe`
4. Download the Linux Miniforge installer from https://github.com/conda-forge/miniforge#miniforge3
5. Change directory within WSL within Windows Terminal by entering `cd <directory where Miniforge installer was downloaded to>`
6. Install Miniforge within WSL within Windows Terminal using `bash Miniforge3-Linux-x86_64.sh`. Once complete it will ask to activate `conda`. Put yes and restart. Enter `wsl.exe` again upon restart of Windows Terminal to be within WSL
7. Create weis environment using
 `conda env create --name weis-env -f https://raw.githubusercontent.com/WISDEM/WEIS/main/environment.yml `
8. Activate environment using `conda activate weis-env` or `source activate weis-env`
9. Update `sudo apt update` 
10. `conda` config --add channels conda-forge`
11. `sudo apt install gcc g++ gfortran libblas-dev liblapack-dev  -y`
12. Clone repository and install (for main branch)
 `git clone https://github.com/WISDEM/WEIS.git`
 `cd WEIS`
 `python setup.py develop` 
13. Some modules may need to be installed. If I remember correctly, I ran the following. I'm not sure if all of these are required.
`conda install wisdem`
`conda install openmdao`
`pip3 install mxnet-mkl==1.6.0 numpy==1.23.1`
14. To run examples, change directory to the 'test' folder using `cd /weis/test`
15. `python run_examples.py`

Hopefully, I remembered everything correctly and this will be useful for others wanting to install on WSL.

closed 2023-10-24 by Russell9798

dzalkind, 2023-10-24:
Thanks for the tips!