NLRWindSystems/WEIS #239: Resolve Package not found error when installing weis

DuncanGWT, 2023-10-19:
## Description
_Describe the bug here.  Please include as many of the sections below as you are able, so we can better assist you._
I am currently trying to install weis, I am on a mac with an M2 chip and already have rosetta installed. At the installation step of updating the environment I am receiving a ResolvePackageNotFound: -smt error. I have tried reinstalling miniconda and it is still not working. Although the steps I am following are not the same as the installation steps, I have followed these steps in the past and successfully installed weis. This is a reinstallation attempt that is not going well.
### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

The installation steps below were provided by Derek Slaughter
1. `conda activate base`
2. `conda create --name weis-env python=3.10`
3. `conda activate weis-env`
4. `conda config --add channels conda-forge`
5. `conda install -y petsc4py mpi4py compilers`
6.  `git clone https://github.com/WISDEM/WEIS.git`
7. `cd WEIS`
8. `git checkout develop`
9. `conda env update --file environment.yml`

### Current behavior
`…conda env update --file environment.yml 
Collecting package metadata (repodata.json): done
Solving environment: failed`

ResolvePackageNotFound: 
  - smt

### Expected behavior
… weis installs and I am able to run the example files

dzalkind, 2023-10-19:
Hi,

It looks like you are having the same issue that Eliot solved in #235.

If you update [these files](https://github.com/WISDEM/WEIS/pull/235/files), it should work.

Best, Dan

dzalkind, 2023-12-22:
Hi Duncan, recently installations with the Mac silicon have been working okay.   We have updated the README installation tips with the latest guidance.

closed 2023-12-22 by dzalkind