NLRWindSystems/WEIS #241: WEIS Error handling/restart after failed Case

mlidGWT, 2023-10-24:
# Description of feature
Hello,

This is more a question about current WEIS capability with the potential to be a feature request if the functionality doesn't already exist. 

I was running a DLC (1.4) using the `openfast_dir` style of example 02, and case 4 out of 40 failed with an error that I don't think is a simple fix/is more endemic to the challenges of the load case. Setting the error aside, I would like to know if there is a way to restart WEIS at the next case in the list (Case 5) and resume until either the end or another case fails. I don't believe restarting the run would fix it, and I would rather skip over the problematic cases for the time being than spend time troubleshooting the error. Is there any way to do this currently?

# Potential solution
One possible way to do this would be for the user to modify the case matrix to remove the completed and problematic Cases, and then run some type of command to get WEIS to resume. Alternatively, if WEIS were to generate all wind and .fst files upfront before running OpenFAST, then I could at least run the remaining .fst files from the command line/a script. A third way might be to have an analysis option to "Skip Fatal", ideally saving the failed case's error message to a log file before moving to the next, and thus allowing all Cases in the matrix to at least be attempted once before the run ends.

dzalkind, 2023-10-24:
Hi Meg,

There is an option to allow failed OpenFAST simulations and continue, like the last option you mention.  Here is an example:

```
General:
    openfast_configuration:
        OF_run_fst: RM1
        # OF_run_dir: outputs/CT_Opt/07_Ptfm_Setup
        save_timeseries: True
        save_iterations: True
        use_exe: True
        allow_fails: True
        fail_value: 9999
```
A comprehensive list of the input options can be found [here in the various schema](https://github.com/WISDEM/WEIS/tree/main/weis/inputs).

Best, Dan

mlidGWT, 2023-10-26:
Hi Dan, 

Thanks for that! I have added it to my file. For reference, options in the modeling_schema don't always work for me (I just provided one example [here](https://github.com/WISDEM/WEIS/discussions/244), and I have others. I'm not sure if there are dependencies or conflicts for some of the options that aren't clear, if I'm doing something wrong, or if some of the options listed don't actually work. 

Best,
Meg

closed 2023-10-26 by mlidGWT