NLRWindSystems/WEIS #248: Error running from develop branch

Russell9798, 2023-10-30:
Hi,

I'm trying to run the examples from the develop branch but I'm getting the following error:

"Found existing potential model: examples/01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi
    - Trying to use this instead of running PyHAMS.
Traceback (most recent call last):
  File "/mnt/c/Users/s2113132/WEIS_develop/examples/06_IEA-15-240-RWT/weis_driver_umaine_semi.py", line 15, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
  File "/mnt/c/Users/s2113132/WEIS_develop/weis/glue_code/runWEIS.py", line 151, in run_weis
    wt_opt.setup(derivatives=False)
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/openmdao/core/problem.py", line 1019, in setup
    model._setup(model_comm, mode, self._metadata)
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/openmdao/core/group.py", line 728, in _setup
    self._setup_procs(self.pathname, comm, mode, self._problem_meta)
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/openmdao/core/group.py", line 633, in _setup_procs
    subsys._setup_procs(subsys.pathname, sub_comm, mode, prob_meta)
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/openmdao/core/component.py", line 222, in _setup_procs
    self.setup()
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/wisdem/commonse/cylinder_member.py", line 2135, in setup
    n_full = self.options["n_full"]
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/openmdao/utils/options_dictionary.py", line 561, in __getitem__
    self._raise(f"Option '{name}' is required but has not been set.")
  File "/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/openmdao/utils/options_dictionary.py", line 245, in _raise
    raise exc_type(full_msg)
RuntimeError: 'towerse_post' <class CylinderPostFrame>: Option 'n_full' is required but has not been set."

FYI, after running `python setup.py develop` I had to run `conda install wisdem` and `conda install pyhams` since I was getting the errors `ModuleNotFoundError: No module named 'wisdem'` and ModuleNotFoundError: No module named 'pyhams'. Trying to run the examples after this resulted in the error above. 

How might I be able to fix this?

Kind regards,

Andrew

gbarter, 2023-11-04:
Hi Andrew,

You have installed some of the dependencies through conda packages instead of from source, which is fine.  However, the underlying WEIS python code has still not been installed correctly.  Can you post the full log from the `python setup.py develop`?

Cheers,
Garrett

Russell9798, 2023-11-08:
Hi Garrett,

This is the log I get from running `python setup.py develop`

`/mnt/c/Users/s2113132/WEIS_develop/setup.py:5: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import run_setup
/mnt/c/Users/s2113132/WEIS_develop/setup.py:7: DeprecationWarning:

  `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
  of the deprecation of `distutils` itself. It will be removed for
  Python >= 3.12. For older Python versions it will remain present.
  It is recommended to use `setuptools < 60.0` for those Python versions.
  For more details, see:
    https://numpy.org/devdocs/reference/distutils_status_migration.html

  from numpy.distutils.command.build_ext import build_ext
['setup.py', 'develop']
Traceback (most recent call last):
  File "/mnt/c/Users/s2113132/WEIS_develop/WISDEM/setup.py", line 104, in <module>
    run_meson_build(staging_dir)
  File "/mnt/c/Users/s2113132/WEIS_develop/WISDEM/setup.py", line 44, in run_meson_build
    raise OSError("The meson command cannot be found on the system")
OSError: The meson command cannot be found on the system
running develop
/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing rosco.egg-info/PKG-INFO
writing dependency_links to rosco.egg-info/dependency_links.txt
writing requirements to rosco.egg-info/requires.txt
writing top-level names to rosco.egg-info/top_level.txt
reading manifest file 'rosco.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'rosco.egg-info/SOURCES.txt'
running build_ext
Creating /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages/rosco.egg-link (link to .)
rosco 2.6.0 is already the active version in easy-install.pth

Installed /mnt/c/Users/s2113132/WEIS_develop/ROSCO
Processing dependencies for rosco==2.6.0
Searching for pandas==2.1.2
Best match: pandas 2.1.2
Adding pandas 2.1.2 to easy-install.pth file

Using /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages
Searching for future==0.18.3
Best match: future 0.18.3
Adding future 0.18.3 to easy-install.pth file
Installing futurize script to /root/miniforge3/envs/weis-env-develop/bin
Installing pasteurize script to /root/miniforge3/envs/weis-env-develop/bin

Using /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages
Searching for PyYAML==6.0.1
Best match: PyYAML 6.0.1
Adding PyYAML 6.0.1 to easy-install.pth file

Using /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages
Searching for scipy==1.11.3
Best match: scipy 1.11.3
Adding scipy 1.11.3 to easy-install.pth file

Using /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages
Searching for pytest==7.4.3
Best match: pytest 7.4.3
Adding pytest 7.4.3 to easy-install.pth file
Installing py.test script to /root/miniforge3/envs/weis-env-develop/bin
Installing pytest script to /root/miniforge3/envs/weis-env-develop/bin

[... 1085 lines cut ...]

Using /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages
Searching for six==1.16.0
Best match: six 1.16.0
Adding six 1.16.0 to easy-install.pth file

Using /root/miniforge3/envs/weis-env-develop/lib/python3.10/site-packages
Finished processing dependencies for WEIS==1.1
INFO:
########### EXT COMPILER OPTIMIZATION ###########
INFO: Platform      :
  Architecture: x64
  Compiler    : gcc

CPU baseline  :
  Requested   : 'min'
  Enabled     : SSE SSE2 SSE3
  Flags       : -msse -msse2 -msse3
  Extra checks: none

CPU dispatch  :
  Requested   : 'max -xop -fma4'
  Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
  Generated   : none
INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /mnt/c/Users/s2113132/WEIS_develop/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py`

gbarter, 2023-11-13:
I see the error:
```
Traceback (most recent call last):
File "/mnt/c/Users/s2113132/WEIS_develop/WISDEM/setup.py", line 104, in
run_meson_build(staging_dir)
File "/mnt/c/Users/s2113132/WEIS_develop/WISDEM/setup.py", line 44, in run_meson_build
raise OSError("The meson command cannot be found on the system")
OSError: The meson command cannot be found on the system
```
That is why WISDEM isn't installing. You can try `conda install wisdem` or if you want to try the installation from source again you could do `conda install meson ninja` and then the `pythons setup.py develop` command again.

closed 2023-11-17 by Russell9798