NLRWindSystems/WEIS #253: Compiling and Running under Apple Silicon Macs, Mamba, and Miniforge3

yonghoonlee, 2023-12-13:
# Description of feature
Describe the feature here and provide some context. Under what scenario would this be useful?

I am currently working on the code with M3 CPU with miniforge3 and mamba environment. The WEIS installation instruction recommends the use of Anaconda x86-64 binary. However, the WEIS (develop branch) seems to work fine with Apple Silicon (macOS arm64) binary version of the miniforge3.

I think the current WEIS is experimenting with mamba instead of conda for its installation and compilation process, so I think it would be good if the installation process could be updated with the Apple Silicon binary version, when mamba is implemented as its primary installation method.

# Potential solution
Can you think of ways to implement this?

A bit more comprehensive testing with Apple Silicon version of virtual environment system would be necessary. With my short experience, it just naturally works. If there is no issue found, installation instruction might need to be updated.

dzalkind, 2023-12-13:
Thanks @yonghoonlee! I think the consensus at NREL is that it's working fine with Apple silicon and mamba/miniforge.  Conda seems to be taking a very long time for many users.  We discussed updating the README yesterday.  @gbarter, is there any reason not to do this yet?

gbarter, 2023-12-14:
No reason to wait.  I just updated the WISDEM [README guidance](https://github.com/WISDEM/WISDEM/?tab=readme-ov-file#installation) yesterday and we can copy paste the relevant sections for the WEIS README too.

yonghoonlee, 2023-12-14:
@gbarter in addition to replacing with miniforge, replacing conda command with mamba would also help a lot for reducing time and conflicts while resolving dependencies. I think major reason to go for miniforge instead of anaconda is to use mamba, but still conda is available there and conda in miniforge still makes issues in resolving dependencies.

gbarter, 2023-12-14:
As of a couple of months ago, miniforge and mambaforge are the same.  Using the `conda` command from miniforge3 and the `mamba` command from mambaforge are aliases of one another.

See the first couple of Q&A at this FAQ:
https://github.com/conda-forge/miniforge?tab=readme-ov-file#whats-the-difference-between-mambaforge-and-miniforge

yonghoonlee, 2023-12-14:
Thank you for clarification. I think I was not updated with the latest info.

gbarter, 2023-12-14:
No worries!

dzalkind, 2023-12-22:
The README on `develop` and `main` have been updated with the latest conda/mamba guidance.

closed 2023-12-22 by dzalkind