NLRWindSystems/WEIS #254: Cannot run examples\02_control_opt\weis_driver_sm.py

Wying301, 2023-12-18:
Hi, when I run _**weis_driver_sm.py**_ (stability_margin example), it seems that the optimization process does not been activated, and the openfast does not been used. I use the default modelling_options file and analysis_options file.

The output is
`(weis-env) MacBook-Air-2:test user$ python test_examples_skinny.py

NOW RUNNING: 02_control_opt/weis_driver_sm

weis_driver_sm /Users/user/WEIS/examples/02_control_opt/weis_driver_sm.py
Using weis.aeroelasticse in ROSCO_toolbox...
6.270707368850708 seconds to run
.
----------------------------------------------------------------------
Ran 1 test in 6.268s
`
I can run _**weis_driver.py**_ and _**weis_driver_loads.py**_ normally. Could anyone tell me why this happens?

dzalkind, 2023-12-18:
Hi,

There are a couple of errors with this example, it seems.  I'm not sure how it passed our tests.

I would skip running this example unless it is critical to your work.

Best, Dan

For later development:

I'll keep this issue open until we can support it.  To fix this example, we'll need to
- Bring in recent fixes to ROSCO that kept up to date with the python-toolbox MBC3 function
- Update or directly use the python-toolbox version of the MBC3 function in WEIS.
- Figure out why tests can pass if the example fails

Wying301, 2023-12-19:
@dzalkind Hi, thank you for your quick reply! As you suggested, I will skip this example and I will also keep a check on this issue.

Wying301, 2023-12-21:
Hi dzalkind,

I have two quick questions about the example “02_loads” by running weis_driver_loads.py. https://github.com/WISDEM/WEIS/tree/main/examples/02_control_opt.

1.       What’s the target of this example? There is no design variables and the optimization is disabled in analysis_options.yaml.

2.     It seems that in each iteration, OpenFAST will run three times by default, so that I can get three sets of files, namely IEA15_0, IEA15_1 and IEA15_2. Why this happen (I mean why three times)? And what’s the entry for this setting?

Thanks,
Ying Wu

答复: WISDEM/WEIS ***@***.***>
日期: 2023年12月19日 星期二 上午12:20
收件人: WISDEM/WEIS ***@***.***>
抄送: Ying Wu ***@***.***>, Author ***@***.***>

dzalkind, 2023-12-21:
Hi,

I updated the documentation in these examples: https://github.com/WISDEM/WEIS/tree/main/examples/02_control_opt

Please let me know if you have further questions.

Best, Dan

Wying301, 2023-12-22:
Hi Dan,

Thank you for providing the detailed description of this example. I think I understand the main idea of the three cases in example/02_control_opt.

If you don’t mind, I also want to ask a few questions about the example/03_NREL5MW_OC3_spar.

1.       I check the Level 3 in modelling_options.yaml file, and the entry “from_openfast” is deactivated. So I think the openfast model here is generated from WISDEM, rather than previously defined like 02_control_opt. My question is there any outputs after the running of WISDEM, like the .fst file or the Cp-Ct-Cq.txt file?

2.       As you suggested, I open the case_matrix.txt file to check the load cases run. I find that the 02_control_opt/weis_driver.py run three load cases (IEA15_0,  IEA15_1,  IEA15_2), while the 03_NREL5MW_OC3_spar/weis_driver.py run only one load case (NREL5MW_OC3_spar_0). My question is where to specify the number and name of load cases to be run? (I check the DLC_driver in modelling_options.yaml, but I cannot understand all entries.)

3.       To run the full optimization, I install NLOpt solvers as described here https://github.com/WISDEM/WEIS/tree/main/examples/03_NREL5MW_OC3_spar. But I still receive the waring “/Users/username/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/visualization/opt_report/opt_report.py:119: DriverWarning:The optimizer report is not applicable for Driver type 'NLoptDriver', which does not support optimization” after the optimization. Why this happens?

4.       In analysis_options.yaml file, the design space for omega_pc in controller is [0.1, 0.5] and for ballast volume is [800, 1500].

-          Before optimization, it says:

Design Vars

{'tune_rosco_ivc.omega_pc': array([0.35]),

    'wisdem.wt.wt_init.floating.memgrp0.ballast_volume': array([900.,   0.])}

-          After optimization, it says:

Objectives

{'wisdem.wt.floatingse.sys.mux.platform_mass': array([7710405.81640525])}
            What’s the meaning of these outputs? It seems that no optimal values for omega_pc and ballast_volume are obtained after optimization.

Sorry to ask such many questions. I am new to the simulation of  FOWT, so some outputs from WEIS is difficult for me to understand. Look forward to your reply!

Thanks,
Ying Wu

答复: WISDEM/WEIS ***@***.***>
日期: 2023年12月22日 星期五 上午1:22

dzalkind, 2023-12-22:
Hi Ying,

We're going to work on more detailed documentation about your questions, as I think they could help more users.  I'll provide some quick answers in the meantime.

1. WISDEM/WEIS output locations are specified in the [analysis options](https://github.com/WISDEM/WEIS/blob/7b05e1924d63fe921f5ce3733b9069a3e9ef0019/examples/03_NREL5MW_OC3_spar/analysis_options.yaml#L2) and openfast locations are specified in the [modeling options](https://github.com/WISDEM/WEIS/blob/7b05e1924d63fe921f5ce3733b9069a3e9ef0019/examples/03_NREL5MW_OC3_spar/modeling_options.yaml#L5).  All inputs to WEIS are defined in the [various schema](https://github.com/WISDEM/WEIS/tree/main/weis/inputs). 

2. DLC inputs are defined in the modeling schema.  We are working on a more detailed documentation of this feature.

3. I think you can ignore this issue.

4. These print statements are openmdao connections in WEIS that are specified as design variables, constraints, or merit figures.  There should be a print statement like this for each iteration.  I'm guessing you only ran one iteration, which you can increase with `max_iter` in the analysis options.

I hope this helps.  More detailed documentation may take some time. These are good questions we can use to document the tool better from the user perspective.  I may break this issue into a few separate issues for easier tracking.

Please let us know what else you find. 

Best, Dan

Wying301, 2023-12-24:
Hi Dan,

Thanks for your reply!

I am trying to run the example 03_NREL5MW_OC3_spar in the full optimization as guided in https://github.com/WISDEM/WEIS/tree/main/examples/03_NREL5MW_OC3_spar and have the following questions:

1.       I don’t find “TMax” and “TStart” under Level 3 – simulation in inputs/modeling_schema.yaml. In fact, when I add these two entries () in modeling_options.yaml., the simulation time does not change too much.

2.       In this case, the controller (omega) and the ballast volume are optimized. While how can we obtain the convergence figures of design variables vs. number of iterations, like that in 02_control_opt, i.e. where to specify the outputs of figures?

3.       I still don’t understand the outputs among iterations. For example, this case run 9 openfast iterations, and the print statements for the last iteration are:

Driver debug print for iter coord: rank0:NLopt_LN_COBYLA|9

----------------------------------------------------------

Design Vars

{'tune_rosco_ivc.omega_pc': array([0.35]),

 'wisdem.wt.wt_init.floating.memgrp0.ballast_volume': array([900.95868573,   0.        ])}

… … (running of WISDEM, OpenFAST, RASCP)… …

Objectives

{'wisdem.wt.floatingse.sys.mux.platform_mass': array([7710405.81640525])}

Does these information mean that the optimal omega is 0.35 and the optimal ballast_volume is 900.95868573, and the corresponding value of objective function (i.e. platform mass) is 7710405.81640525?

Except for above questions, there may exist a few typos:

-      /WEIS/weis/inputs/analysis_schema.yaml: line 399 and line 411, the descriptions of Max_TwrBsMyt and DEL_TwrBsMyt seem not correct.

-      For 03_NREL5MW_OC3_spar, if try to run optimization procedure, should the flags for design_variables in analysis_options.yaml be converted to True so that these variables can be optimized.

Thanks for your attention to my previous emails! And Merry Christmas～

Thanks,
Ying

答复: WISDEM/WEIS ***@***.***>
日期: 2023年12月23日 星期六 上午1:01

dzalkind, 2023-12-26:
`TMax` and `TStart` are some of the input options that change with DLC.  You can set them with `analysis_time` and `transient_time` in the modeling options: https://github.com/WISDEM/WEIS/blob/7b05e1924d63fe921f5ce3733b9069a3e9ef0019/examples/03_NREL5MW_OC3_spar/modeling_options.yaml#L108

An example notebook for reviewing optimization convergence can be found here: https://github.com/WISDEM/WEIS/blob/main/examples/16_postprocessing/rev_Opt.ipynb

Wying301, 2024-01-11:
Hi Dan,

I am working on the DLCs analysis example in WEIS/examples/15_RAFT_Studies and I have the following questions.

1.       In the modelling_options_level3.yaml, many DLCs are specified. I run them one by one separately to test whether they can be run normally.

-      For DLC ‘1.1’, ‘1.3’, ‘1.6’, ‘6.1’,’6.3’, the OpenFAST failed to run due to the error “FAST_InitializeAll:ED_Init:ED_ReadInput:ReadTowerFile:TwSSM2Sh: value is not a finite real number.” I also reported this issue here https://github.com/WISDEM/WEIS/issues/255, with more information.

-      For DLC ‘1.4’ and ‘1.5’, the DLC cases failed to run, with the following errors:

======================================================================

FAIL: test_all_scripts (__main__.TestSkinnyExamples) [Running: 15_RAFT_Studies/weis_driver_dlc_level3] (i=0)

----------------------------------------------------------------------

Traceback (most recent call last):

  File "/Users/wuying/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/system.py", line 2668, in _call_user_function

    yield

  File "/Users/wuying/anaconda3/envs/weis-env/lib/python3.10/site-packages/openmdao/core/explicitcomponent.py", line 289, in _compute_wrapper

    self.compute(self._inputs, self._outputs,

  File "/Users/wuying/WEIS/RAFT/raft/omdao_raft.py", line 646, in compute

    if np.iscomplex(results['response'][name]).any():

KeyError: 'response'

  1.  When comparing the DLCs, I find the DLCs ‘6.1’ and ‘6.3’ only has one wind speed (70m/s and 56m/s), while other DLCs have wind speed ranging from 3m/s to 25m/s with an interval of 2m/s. Where can I find the detailed definitions of these IEA DLCs?

Look forward to your reply!

Thanks,

Ying

答复: WISDEM/WEIS ***@***.***>
日期: 2023年12月27日 星期三 上午1:14

dzalkind, 2024-07-23:
This should be fixed with the latest release.  Please re-open if you still have issues.

closed 2024-07-23 by dzalkind