NLRWindSystems/WEIS #258: OpenFAST Simulation Aborted While Running 'weis_driver.py' in 'WEIS/examples/13_DTQP' Directory

saeidb71, 2024-01-10:
I'm utilizing the 'develop' branch and attempted to execute 'weis_driver.py' within the 'WEIS/examples/13_DTQP' directory. However, it resulted in the following error:

"Running ElastoDyn.
The Nodal outputs section of the ElastoDyn input file is either missing or improperly formatted.
FAST_InitializeAll:ED_Init:ED_ReadInput:ReadTowerFile:TwSSM2Sh: the value is not a valid finite real number.
OpenFAST encountered a critical error during module initialization.
Simulation error severity: FATAL ERROR
Aborting OpenFAST."

dzalkind, 2024-01-10:
Try adding this modeling option to the one you are using: https://github.com/WISDEM/WEIS/blob/7b05e1924d63fe921f5ce3733b9069a3e9ef0019/examples/06_IEA-15-240-RWT/modeling_options_umaine_semi.yaml#L27

saeidb71, 2024-01-10:
Thanks. The issue has been resolved. I will go ahead and close the issue now.

dzalkind, 2024-01-10:
I'm going to re-open it so I actually fix it when we go through and clean up the examples soon.

saeidb71, 2024-01-11:
Oh, I see. In that case, you may need to also change OF_run_dir. It is currently set to:

OF_run_dir: /Users/dzalkind/Tools/WEIS-4/outputs/13_DTQP_3

closed 2024-07-23 by dzalkind