NLRWindSystems/WEIS #263: User Defined control optimization based on 02_control_opt

wind2021, 2024-02-01:
Hello, everyone!

After a period of learning and debugging, I have been able to run basic cases. But when I started user Defined control optimization based on 02_control_opt, something went wrong,  although I think I have already referred to the relevant code.
1. When I add **wave_spectrum** at **DLCs** in **DLC_driver** of _modeling_options.yaml_, I used the below words:
wave_spectrum: ["JONSWAP", "unit", "JONSWAP"]
which refers to the statement of  **line 158** of https://github.com/WISDEM/WEIS/blob/main/examples/05_IEA-3.4-130-RWT/modeling_options.yaml
But it seems not work, and It has an error message, which tell me there's something wrong with this expression.
2. I also want to know if the expression below is right or not, especially **Z0**:
          turbulent_wind:
              HubHt: 90
              RefHt: 90
              URef: 8
              GridHeight: 160
              GridWidth: 160
              IECturbc: 0.05
              PLExp: 0.14
              Z0: "0.03"
3. Lastly, could you give me some advice on In below-rated operation? As I want that ROSCO uses a wind speed estimate to determine the optimal generator speed set point for tracking the optimal tip speed ratio. What about the setting below for ROSCO in _modeling_options.yaml_ and **design_variables/merit_figure/constraints** for _analysis_options.yaml_?
ROSCO:
    flag: True
    SD_Mode: 0
    PS_Mode: 0
    ps_percent: 0.85
    F_LPFType: 2
    F_NotchType: 2
    Fl_Mode: 2
    tuning_yaml: /home/wind/WEIS/ROSCO/Tune_Cases/NREL5MW.yaml
    zeta_pc: [ 0 ]
    omega_pc: [ 0 ]
    U_pc: [ 8 ]
    zeta_vs: 0.85                     # Torque controller desired damping ratio [-]
    omega_vs: 0.12
    twr_freq: 3.2
    ptfm_freq: 0.2
    Kp_float: -10

design_variables:
  control:
    servo:
      torque_control:
        omega:
          flag: True
          min: 0.1
          max: 1.0
        zeta:
          flag: True
          min: 0.1
          max: 1.5
merit_figure: Cp  # Merit figure of the optimization problem. The options are 'AEP' - 'LCOE' - 'Cp' - 'blade_mass' - 'blade_tip_deflection'
constraints:
  control:
    nacelle_acceleration:
      flag: false
      max: 3.0

If anyone can give me some kindly remind, I would be very grateful. Thank you all.
[image]
[image]
[image]
[modeling and analysis files.zip](https://github.com/WISDEM/WEIS/files/14122081/modeling.and.analysis.files.zip)

dzalkind, 2024-02-06:
Hi,

Thank you for your patience.  Let me try to answer all your questions, or at least point you in the proper direction towards solving them.

1. That case is commented out, it does not appear to affect the OpenFAST inputs when I search for `wave_spectrum` in the code.  Either way, it should be the same length as the number of wind speed you suggest.
2. This one is a bit tricker, it may be supported.  There should be a `wind/` directory where your OpenFAST files are generated.  There, you can see if the input is affecting the wind file.
3. What are you trying to achieve with the torque control parameters?  The Cp will not be affected by the torque control inputs.  `AEP` should work. I would not set `omega_pc` and `zeta_pc` to 0.  I'm guessing it throws some kind of error in that case.

The latest descriptions of the WEIS inputs can be found here: https://github.com/WISDEM/WEIS/tree/docs/docs/inputs

Generally, straying from the given set of examples requires some familiarity with the source code and potentially making changes.

I hope this helps.

Best, Dan

wind2021, 2024-02-08:
Thank you so much, Dan!
Sorry for the late reply.

1. I have tried many words for wave_spectrum, but all failed. So do you have ever use this in WEIS? Can anyone who had use this give me some help?
2. The line 119 works well. 
3.  I followed your advice, and use AEP, but it seems here are some error with runFAST_pywrapper.py. Is this a bug or I miss something? What do you think?
[image]

dzalkind, 2024-02-08:
Hi,

1. The code is not configured to use `wave_spectrum` as a DLC input at this point.  You will need to make edits in the source code to enable that feature.
3. Is there any more information about why OpenFAST failed?  It's likely not because of the chosen `AEP` merit figure, but something else in the ROSCO or OpenFAST modeling options.  [Example 03 in this branch is configured for the NREL 5MW](https://github.com/WISDEM/WEIS/blob/docs/examples/03_NREL5MW_OC3_spar/modeling_options.yaml).

Dan

dzalkind, 2024-07-23:
I'm closing this for now.  Please re-open if you have further questions/issues that are unresolved.

closed 2024-07-23 by dzalkind