NLRWindSystems/WEIS #269: Running Multiple OpenFAST Simulations -  ROSCO_IO.f90 already opened in another unit

saeidb71, 2024-02-18:
I have two .fst files, namely "NREL5MW_OC4_semi_wind_0.fst" and "NREL5MW_OC4_semi_wind_1.fst", along with their corresponding .dat files. How can I run these two cases? I tried using the method from "01_aeroelasticse", but using "case_inputs", I couldn't add both .fst files. So, I attempted to run the first file and then the second one, but it generated this error:

"At line 851 of file /Users/runner/miniforge3/conda-bld/rosco_1706552890583/work/rosco/controller/src/ROSCO_IO.f90 (unit = 14)
Fortran runtime error: File already opened in another unit"

How can I resolve this issue?

saeidb71, 2024-02-18:
Update:

I found out that by using fastBatch.run_multi(1), I could run the first simulation and then the second one. However, fastBatch.run_serial() creates the error I mentioned previously.

dzalkind, 2024-02-19:
Hi Saeid,

I haven't seen this error before.  The `01_aeroelasaticse` scripts take a single base file and runs different configurations of that template, where the `case_inputs` determine which parameters change.

I'm not quite sure exactly what you're doing, so perhaps push a simple working example to a WEIS fork of yours and I can help debug.

Best, Dan

closed 2024-07-23 by dzalkind