NLRWindSystems/WEIS #270: PermissionError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), [WinError 5] Access is denied

edpeca, 2024-02-19:
c:\repo\NREL\WEIS\examples\04_NREL5MW_OC4_semi\
python weis_driver.py

-------------------

Using weis.aeroelasticse in rosco.toolbox...

=========
wisdem.wt
=========
←[00m←[00m←[00m←[00m←[00m←[00m←[00mWARNING: Frequency index 0 has been used again for i=0 in the x-direction
WARNING: Frequency index 3 has been used again for i=1 in the y-direction
WARNING: Frequency index 1 has been used again for i=2 in the z-direction
RuntimeWarning: C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\wisdem\commonse\utilization_dnvgl.py:316
The number of calls to function has reached maxfev = 50.NL: NLBGS 1 ; 2.44948974e+30 1
←[00m←[00m←[00m←[00m←[00m←[00m←[00mWARNING: Frequency index 0 has been used again for i=0 in the x-direction
WARNING: Frequency index 3 has been used again for i=1 in the y-direction
WARNING: Frequency index 1 has been used again for i=2 in the z-direction
RuntimeWarning: C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\wisdem\commonse\utilization_dnvgl.py:316
The number of calls to function has reached maxfev = 50.NL: NLBGS 2 ; 8.59661346e+09 3.50955275e-21
NL: NLBGS Converged
ORBIT library intialized at 'C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\wisdem\library'
########################################
Objectives
Turbine AEP: 23.9496905660 GWh
Blade Mass:  16469.8925506885 kg
LCOE:        146.3620584624 USD/MWh
Tip Defl.:   4.5800216220 m
########################################
Cp-Ct-Cq surfaces completed at 20 %
Cp-Ct-Cq surfaces completed at 40 %
Cp-Ct-Cq surfaces completed at 60 %
Cp-Ct-Cq surfaces completed at 80 %
Cp-Ct-Cq surfaces completed at 100 %
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox
-----------------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2620, in _call_user_function
    yield
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 283, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\NREL\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 605, in compute
    summary_stats, extreme_table, DELs, Damage, case_list, case_name, chan_time, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\NREL\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 1835, in run_FAST
    WindFile_type[i_case] , WindFile_name[i_case] = generate_wind_files(
                                                    ^^^^^^^^^^^^^^^^^^^^
  File "C:\NREL\WEIS\weis\aeroelasticse\turbsim_util.py", line 260, in generate_wind_files
    wrapper.execute()
  File "C:\NREL\WEIS\weis\aeroelasticse\turbsim_util.py", line 220, in execute
    subprocess.call(exec_string)
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\subprocess.py", line 1538, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [WinError 5] Access is denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\NREL\WEIS\examples\04_NREL5MW_OC4_semi\weis_driver.py", line 16, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\NREL\WEIS\weis\glue_code\runWEIS.py", line 199, in run_weis
    wt_opt.run_model()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\problem.py", line 640, in run_model
    self.model.run_solve_nonlinear()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 4416, in run_solve_nonlinear
    self._solve_nonlinear()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\group.py", line 3180, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
    ^^^^^^^^^^^^
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\solver.py", line 800, in _gs_iter
    subsys._solve_nonlinear()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 312, in _solve_nonlinear
    self._compute_wrapper()
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 266, in _compute_wrapper
    with self._call_user_function('compute'):
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\contextlib.py", line 158, in __exit__
    self.gen.throw(typ, value, traceback)
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2626, in _call_user_function
    raise err_type(
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2620, in _call_user_function
    yield
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 283, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\NREL\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 605, in compute
    summary_stats, extreme_table, DELs, Damage, case_list, case_name, chan_time, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\NREL\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 1835, in run_FAST
    WindFile_type[i_case] , WindFile_name[i_case] = generate_wind_files(
                                                    ^^^^^^^^^^^^^^^^^^^^
  File "C:\NREL\WEIS\weis\aeroelasticse\turbsim_util.py", line 260, in generate_wind_files
    wrapper.execute()
  File "C:\NREL\WEIS\weis\aeroelasticse\turbsim_util.py", line 220, in execute
    subprocess.call(exec_string)
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\ProgramData\anaconda3\envs\weis-env\Lib\subprocess.py", line 1538, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), [WinError 5] Access is denied

edpeca, 2024-02-19:
Yes, I am currently using develop branch. I believe it might be due to Windows Secutiry blocking the access to OpenFast
[image]

edpeca, 2024-02-19:
[image]

gbarter, 2024-02-19:
That was going to be my next question.  If you just type `openfast -h` and `turbsim -h` at the command line, does that work?  Sounds like that is blocked?

edpeca, 2024-02-19:
`openfast -h ` works, but  `turbsim -h` prompts "Acced denied" and it is blocked by Windows Secutiry. I have raised a ticket in our IT management system

gbarter, 2024-02-19:
Sounds good.  Please let us know if the Windows Security fix resolves the issue.

closed 2024-07-23 by dzalkind