NLRWindSystems/WEIS #272: develop branch installation process fails with nlopt

yonghoonlee, 2024-02-27:
## Description
_Describe the bug here_

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. I followed the exact installation instruction under miniforge3 environment on macOS (Apple Silicon)
2. conda installations were all successful
3. pip install -e . process fails with the following message:

```
  Building wheel for nlopt (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [58 lines of output]
      running bdist_wheel
      running build
      running build_ext
      Traceback (most recent call last):
        File "/private/var/folders/tr/0d2kxxl51_g_zgbgjmqwmv6c0000gp/T/pip-install-op97d8v9/nlopt_6bc78d3d85c24627a42e54740f5e50d0/setup.py", line 20, in run
          subp.check_call(["cmake", "--version"])
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 408, in check_call
          retcode = call(*popenargs, **kwargs)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 389, in call
          with Popen(*popenargs, **kwargs) as p:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 1026, in __init__
          self._execute_child(args, executable, preexec_fn, close_fds,
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 1953, in _execute_child
          raise child_exception_type(errno_num, err_msg, err_filename)
      FileNotFoundError: [Errno 2] No such file or directory: 'cmake'
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/tr/0d2kxxl51_g_zgbgjmqwmv6c0000gp/T/pip-install-op97d8v9/nlopt_6bc78d3d85c24627a42e54740f5e50d0/setup.py", line 85, in <module>
          setup(
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/wheel/bdist_wheel.py", line 368, in run
          self.run_command("build")
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/Users/yonghoonlee/miniforge3/envs/weis-env/lib/python3.11/site-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
[... 278 lines cut ...]
treon                     0.1.4              pyhd8ed1ab_0    conda-forge
types-python-dateutil     2.8.19.20240106    pyhd8ed1ab_0    conda-forge
typing-extensions         4.10.0               hd8ed1ab_0    conda-forge
typing_extensions         4.10.0             pyha770c72_0    conda-forge
typing_utils              0.1.0              pyhd8ed1ab_0    conda-forge
tzdata                    2024a                h0c530f3_0    conda-forge
unixodbc                  2.3.12               h0e2417a_0    conda-forge
uri-template              1.3.0              pyhd8ed1ab_0    conda-forge
urllib3                   2.2.1              pyhd8ed1ab_0    conda-forge
wcwidth                   0.2.13             pyhd8ed1ab_0    conda-forge
webcolors                 1.13               pyhd8ed1ab_0    conda-forge
webencodings              0.5.1              pyhd8ed1ab_2    conda-forge
websocket-client          1.7.0              pyhd8ed1ab_0    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
widgetsnbextension        4.0.10             pyhd8ed1ab_0    conda-forge
wisdem                    3.13.0          nompi_h191ec86_103    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
zeromq                    4.3.5                h965bd2d_0    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
```

dzalkind, 2024-02-27:
Hi Yong Hoon,

Please try `pip install -e . --no-deps` to get going.  We should probably update the installation instructions, too.

Dan

dzalkind, 2024-02-27:
You may also need to purge the pip cache with `pip cache purge`

yonghoonlee, 2024-02-27:
> Hi Yong Hoon,
> 
> Please try `pip install -e . --no-deps` to get going. We should probably update the installation instructions, too.
> 
> Dan

This resolved the issue. Thank you!

closed 2024-02-27 by yonghoonlee