NLRWindSystems/WEIS #276: Recently DOE feature is broken

yonghoonlee, 2024-04-08:
## Description

DOE studies are not running anymore. I tried to track down when and where it was broken, but I could not. So, I think I will need some help. No matter what options (FullFact, LatinHypercube, etc) provided, WEIS only runs for a single case and stops. I tried with example 09 and customized DOE input with example 15 as well. No luck with all of them.

It might not be the problem of WEIS itself. I checked recent commit histories, and could not find something that may break DOE feature. It could be issues of conda-installed dependencies. But, I could not find clear clue on anything related to that yet.

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Using miniforge3, installed WEIS develop branch with instructions included.
2. Run example 09

### Current behavior

It runs only one case and stops, no matter what DOE options provided.

### Expected behavior

It should compute many cases with varying input values.

### Code versions
```
# packages in environment at /home/yhlee/miniforge3/envs/weis-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
anyio                     4.3.0              pyhd8ed1ab_0    conda-forge
argon2-cffi               23.1.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0          py311h459d7ec_4    conda-forge
arrow                     1.3.0              pyhd8ed1ab_0    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
async-lru                 2.0.4              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
babel                     2.14.0             pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.12.3             pyha770c72_0    conda-forge
bleach                    6.1.0              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
brotli-python             1.1.0           py311hb755f60_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hb3a22ac_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
comm                      0.2.2              pyhd8ed1ab_0    conda-forge
contourpy                 1.2.1           py311h9547e67_0    conda-forge
control                   0.10.0             pyhd8ed1ab_0    conda-forge
curl                      8.7.1                hca28451_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
dearpygui                 1.11.1                   pypi_0    pypi
debugpy                   1.8.1           py311hb755f60_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
dill                      0.3.8              pyhd8ed1ab_0    conda-forge
docopt                    0.6.2                      py_1    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
et_xmlfile                1.1.0              pyhd8ed1ab_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.6.2                h59595ed_0    conda-forge
[... 290 lines cut ...]
xcb-util                  0.4.0                hd590300_1    conda-forge
xcb-util-image            0.4.0                h8ee46fc_1    conda-forge
xcb-util-keysyms          0.4.0                h8ee46fc_1    conda-forge
xcb-util-renderutil       0.3.9                hd590300_1    conda-forge
xcb-util-wm               0.4.1                h8ee46fc_1    conda-forge
xkeyboard-config          2.41                 hd590300_0    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.1.1                hd590300_0    conda-forge
xorg-libsm                1.2.4                h7391055_0    conda-forge
xorg-libx11               1.8.9                h8ee46fc_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.5                h59595ed_1    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
```

yonghoonlee, 2024-04-08:
I received conda environment file from a colleague who has venv that works with DOE functionality. With a small modification, this actually worked for me. However, I was unable to figure out what dependency makes it breaking. It was not OpenMDAO. I tried multiple previous versions of OpenMDAO and it did not have any effect.

Here, I am attaching the working environment yaml file:

```
name: weis-env
channels:
  - conda-forge
  - defaults
dependencies:
  - ampl-mp=3.1.0=hbec66e7_1006
  - anyio=4.3.0=pyhd8ed1ab_0
  - appnope=0.1.4=pyhd8ed1ab_0
  - argon2-cffi=23.1.0=pyhd8ed1ab_0
  - argon2-cffi-bindings=21.2.0=py311heffc1b2_4
  - arrow=1.3.0=pyhd8ed1ab_0
  - asttokens=2.4.1=pyhd8ed1ab_0
  - async-lru=2.0.4=pyhd8ed1ab_0
  - attrs=23.2.0=pyh71513ae_0
  - babel=2.14.0=pyhd8ed1ab_0
  - beautifulsoup4=4.12.3=pyha770c72_0
  - bleach=6.1.0=pyhd8ed1ab_0
  - brotli=1.1.0=hb547adb_1
  - brotli-bin=1.1.0=hb547adb_1
  - brotli-python=1.1.0=py311ha891d26_1
  - bzip2=1.0.8=h93a5062_5
  - c-ares=1.26.0=h93a5062_0
  - ca-certificates=2024.2.2=hf0a4a13_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - certifi=2024.2.2=pyhd8ed1ab_0
  - cffi=1.16.0=py311h4a08483_0
  - charset-normalizer=3.3.2=pyhd8ed1ab_0
  - colorama=0.4.6=pyhd8ed1ab_0
  - comm=0.2.1=pyhd8ed1ab_0
  - contourpy=1.2.0=py311hd03642b_0
  - control=0.9.4=pyhd8ed1ab_0
  - curl=8.5.0=h2d989ff_0
  - cycler=0.12.1=pyhd8ed1ab_0
  - debugpy=1.8.1=py311h92babd0_0
  - decorator=5.1.1=pyhd8ed1ab_0
  - defusedxml=0.7.1=pyhd8ed1ab_0
  - dill=0.3.8=pyhd8ed1ab_0
  - docopt=0.6.2=py_1
  - entrypoints=0.4=pyhd8ed1ab_0
  - et_xmlfile=1.1.0=pyhd8ed1ab_0
  - exceptiongroup=1.2.0=pyhd8ed1ab_2
  - executing=2.0.1=pyhd8ed1ab_0
  - fatpack=0.7.7=pyhd8ed1ab_0
  - fonttools=4.48.1=py311h05b510d_0
  - fqdn=1.5.1=pyhd8ed1ab_0
  - freetype=2.12.1=hadb7bae_2
  - gettext=0.21.1=h0186832_0
  - git=2.43.0=pl5321h015987d_1
  - h11=0.14.0=pyhd8ed1ab_0
  - h2=4.1.0=pyhd8ed1ab_0
  - hpack=4.0.0=pyh9f0ad1d_0
  - httpcore=1.0.3=pyhd8ed1ab_0
  - httpx=0.26.0=pyhd8ed1ab_0
  - hyperframe=6.0.1=pyhd8ed1ab_0
  - idna=3.6=pyhd8ed1ab_0
  - importlib-metadata=7.0.1=pyha770c72_0
  - importlib_metadata=7.0.1=hd8ed1ab_0
  - importlib_resources=6.1.1=pyhd8ed1ab_0
  - iniconfig=2.0.0=pyhd8ed1ab_0
  - ipopt=3.14.14=h6639f4a_1
  - ipykernel=6.29.2=pyh3cd1d5f_0
  - ipython=8.21.0=pyh707e725_0
  - ipywidgets=8.1.2=pyhd8ed1ab_0
  - isoduration=20.11.0=pyhd8ed1ab_0
[... 172 lines cut ...]
  - typing_extensions=4.9.0=pyha770c72_0
  - typing_utils=0.1.0=pyhd8ed1ab_0
  - tzdata=2024a=h0c530f3_0
  - unixodbc=2.3.12=h0e2417a_0
  - uri-template=1.3.0=pyhd8ed1ab_0
  - urllib3=2.2.1=pyhd8ed1ab_0
  - wcwidth=0.2.13=pyhd8ed1ab_0
  - webcolors=1.13=pyhd8ed1ab_0
  - webencodings=0.5.1=pyhd8ed1ab_2
  - websocket-client=1.7.0=pyhd8ed1ab_0
  - wheel=0.42.0=pyhd8ed1ab_0
  - widgetsnbextension=4.0.10=pyhd8ed1ab_0
  - wisdem=3.13.0=nompi_h191ec86_103
  - xorg-libxau=1.0.11=hb547adb_0
  - xorg-libxdmcp=1.1.3=h27ca646_0
  - xz=5.2.6=h57fd34a_0
  - yaml=0.2.5=h3422bc3_2
  - zeromq=4.3.5=h965bd2d_0
  - zipp=3.17.0=pyhd8ed1ab_0
  - zlib=1.2.13=h53f4e23_5
  - zstd=1.5.5=h4f39d0f_0
  - pip:
      - dearpygui
      - marmot-agents==0.2.5
```

dzalkind, 2024-04-10:
One of the major differences is wisdem (3.13 vs. 3.14).

I know we fiddled with the DOE test recently, @ptrbortolotti. Do you have any ideas or directions to point @yonghoonlee on the WISDEM side?

ptrbortolotti, 2024-04-10:
I can reproduce the error, I'll work to fix it asap

ptrbortolotti, 2024-04-10:
I'm afraid it will take a wisdem release to fix this. @gbarter and I will work on this

ptrbortolotti, 2024-04-10:
@yonghoonlee , if you can, please try this WISDEM branch https://github.com/WISDEM/WISDEM/tree/compare_designs

dzalkind, 2024-04-10:
Here are some tips on using a non-conda installed WISDEM: https://github.com/WISDEM/WEIS/blob/develop/docs/how_to_contribute_code.rst#developing-python-modules-used-in-weis

yonghoonlee, 2024-04-11:
@ptrbortolotti It seems that the `compare_designs` branch made DOE working again. I will more thoroughly testing with other case studies soon.

@dzalkind Thank you for your help on managing non-conda WISDEM installation.

dzalkind, 2024-07-23:
@yonghoonlee, has this issue been resolved?

yonghoonlee, 2024-07-23:
Yes I will close this issue. Thanks

closed 2024-07-23 by yonghoonlee