NLRWindSystems/WEIS #277: Installation instructions on 'main' points to environment.yaml in the 'develop' branch.

mayankchetan, 2024-04-16:
## Description
Installation instructions on `main` points to environment.yaml in the `develop` branch.

### Steps to reproduce issue
I have not come across an issue, but could lead to potential incopatabilties when locking library version in Develop

1. Install WEIS following instructions on README.
2. Do not change branch from main.

eg: these updates in `develop` (https://github.com/WISDEM/WEIS/blob/394d9223dea784bac68eca7fc9be9f846516f841/environment.yml#L40) could possibly break the WEIS install using main.

### Current behavior
Installs libraries from the `develop` branch, when using `main`.

https://github.com/WISDEM/WEIS/blob/394d9223dea784bac68eca7fc9be9f846516f841/README.md?plain=1#L50

### Expected behavior
The instructions to install WEIS points to the `environment.yaml` within the same branch.

Instructions could be re-arranged to clone the git repo, and change branch before creating the conda env using the locally available `environment.yaml` file

### Code versions
- Not applicable

mayankchetan, 2024-04-17:
@yonghoonlee, This is a great idea! @dzalkind and I briefly  spoke about something similar. Going with the assumption that everyone has GIT isnt that bad. The BASH warning when the user doesn't have GIT should point them in the right direction.

gbarter, 2024-04-17:
Small comment that you shouldn't assume `git` is present already, especially on windows.  That is one reason to start with the conda environment.

dzalkind, 2024-07-23:
Fixed in latest release.

closed 2024-07-23 by dzalkind