NLRWindSystems/WEIS #282: Example 1: os.PathLike object, NoneType error

amineeva-GWT, 2024-05-20:
## Description
When attempting to run example 1, I kept getting the following error:

`RuntimeError: Error 4: b'FAST_InitializeAll:HydroDyn_Init:Waves_Init:VariousWaves_Init: The random number generator in use differs from the original code provided by NREL. This pRNG uses 8 seeds instead of the 2 in the HydroDyn input file.\nFAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library /Users/amineeva/WEIS/local/lib/libdiscon.dylib could not be loaded. Check that the file exists in the specified location and that it is compiled for 64-bit applications.`

This seems to be an issue with how WEIS is interfacing with ROSCO - ultimately, the DLL does not point to the correct file, and the example can not run as a result. 

### Steps to reproduce issue
1. Clone the main branch of the WEIS repository onto your computer
2. Open terminal, enter the weis environment
3. Enter the examples folder within the WEIS repository
4. Enter example 01 within the examples folder
5. Attempt to run file 'run_general.py' from the examples folder. I did this using the line `python run_general.py` from the terminal line. 

### Current behavior
Currently, running this does not result in a successful run of the example. Instead, the example has a runtime error (and a random number generator error) that comes from the dynamic library not being found. It seems like this location and library is not creating when running the example, so the file is trying to get to an address that doesn't exist. 

### Expected behavior
It should run example 1 (without error). I am newly learning WEIS and do not know what example 1 should have looked like, but after fixing the problem it seems to be running a time series loop that is meant to simulate ROSCO tuning the controller values. 

### Code versions
_List versions only if relevant_
- main branch, commit: https://github.com/WISDEM/WEIS/commit/9ec2751c5ad23941512400c62536b997c8647a80

gbarter, 2024-05-28:
Hello, it looks like you've edited the ServoDyn entry to be a local path?  If you haven't done that, please do so and double check that the `libdiscon` file is truly there.  Also, the ROSCO library should be appropriately compiled for your machine.  If you are new to OpenFAST and ROSCO, and these are unfamiliar suggestions, please let me know.

If it helps, we are trending to a release of WEIS, so working off of the `develop` branch might not be a bad idea if you are comfortable with that.

dzalkind, 2024-07-23:
I think this issue has been resolved with the fix in WISDEM.  If not, please re-open and let us know how we can help.