NLRWindSystems/WEIS #283: Example 6: SS and PS

amineeva-GWT, 2024-05-20:
## Description
When attempting to run example 6 within the terminal, I got a runtime error because variable related to blade geometry (spar_cap_ss and spar_cap_ps) were not able to be connected. 

### Steps to reproduce issue
1. Clone the main branch of the WEIS repository onto your computer
2. Open terminal, enter the weis environment
3. Enter the examples folder within the WEIS repository
4. Enter example 06 within the examples folder
5. Attempt to run file 'runWEIS.py' from the examples folder. I did this using the line `python runWEIS.py` from the terminal line.

### Current behavior
[image]

### Expected behavior
Not sure what example 6 is meant to run, but it should be running without error. The various `blade.opt_var` look like they exist in the backend but haven't quite made it into the geometry.yaml file, which is why I think they aren't being "connected."

### Code versions
_List versions only if relevant_
- main branch, commit: https://github.com/WISDEM/WEIS/commit/9ec2751c5ad23941512400c62536b997c8647a80

gbarter, 2024-05-28:
Can you give this a try on the `develop` branch?  That has been heavily tested recently and all should be working.  I have a suspicion that you might have to start over with a new environment though as I think part of the problem is that the `main` branch has hard fixed version requirements that are causing this error and other conflicts.

dzalkind, 2024-07-23:
I think this issue has been resolved in the latest release.  If not, please re-open it, and we'd be happy to help.

closed 2024-07-23 by dzalkind