NLRWindSystems/WEIS #286: 9 Character Limit for Outputs

amineeva-GWT, 2024-06-05:
## Description
I'm successfully running an OpenFast simulation using WEIS, but am only getting data from some outputs I list in the InflowFile.dat file. Specifically, in the InflowFile.dat file it seems that any output variables with names longer than 9 characters are somehow not being read by WEIS. However, they show up in the .out files when I run the same .fst file directly through OpenFast. 

### Steps to reproduce issue
1. Set up a basic WEIS model with modeling.yaml, analysis.yaml, geometry.yaml, and weis_driver.py files. Make sure that the files are referencing each other correctly.
2. Create/generate a test.fst file that points to a tester_InflowFile.dat.
3. Within the tester_InflowFile.dat, list the following output parameters:
"Wind1VelX"
"Wind1VelY"
"Wind1VelZ"
"WindHubVelX"
"WindHubVelXY"
"WindDiskVelXY"
"WindHubVelMag"
"WindDiskVelMag"
"WindHubAngXY"
"WindDiskAngXY"
"WindMeas1"
5. Set the modeling.yaml to Level 3 and point towards test.fst. 
6. Run weis_driver.py 
7. Open any .out file - you should only be able to see output parameters "Wind1VelX", "Wind1VelY", "Wind1VelZ", and "WindMeas1". Alternatively, you could also look at the generated copy of the InflowFile.dat within your outputs folder - the same should be seen there. 
8. Optional - if you run the test.fst file directly with OpenFast, all output parameters should be present. 

### Current behavior
This seems like a WEIS issue rather than an OpenFast one because I'm getting outputs when running directly through OpenFast. Something to note is that these variables have names that are longer than 9 characters, which goes against the output channel rules listed in the following outlist parameter document: https://view.officeapps.live.com/op/view.aspx?src=https%3A%2F%2Fopenfast.readthedocs.io%2Fen%2Fmain%2F_downloads%2F3f19498a5dc774461e022b671ff01ec6%2FOutListParameters.xlsx&wdOrigin=BROWSELINK 

### Expected behavior
Hopefully, WEIS should recognize and track all outputs.

amineeva-GWT, 2024-06-06:
Yes, adding them directly to the dictionary creates outputs for those variables. Thanks for your help!

dzalkind, 2024-06-06:
Thanks for following up!  I'll re-open this until we implement the more permanent fix.

dzalkind, 2024-10-02:
@mayankchetan , can we close this since OpenFAST will handle I/O writing soon?

mayankchetan, 2024-10-02:
Yes, we expect to release `openfast_io` with the v4.0 release and then I'll have it integrated into WEIS soon after

closed 2024-11-08 by dzalkind