NLRWindSystems/WEIS #295: ImportError: cannot import name 'validate_without_defaults' from 'wisdem.inputs'

TheMercer, 2024-07-09:
## Description
weis validation script, trying to reference scripts that dont exist in the repo. 

This was working on my old laptop (dont have access to anymore), fresh install now have a problem
### Steps to reproduce issue
I followed install instructions given on github

1. …
2. …
3. …

### Current behavior

>python weis_driver_same.py
Traceback (most recent call last):
  File "~\weis_driver_same.py", line 5, in <module>
    from weis.glue_code.runWEIS     import run_weis
  File "C:\Users\ksm\WEIS\weis\glue_code\runWEIS.py", line 4, in <module>
    from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
  File "C:\Users\ksm\WEIS\weis\glue_code\gc_LoadInputs.py", line 6, in <module>
    import weis.inputs as sch
  File "C:\Users\ksm\WEIS\weis\inputs\__init__.py", line 1, in <module>
    from weis.inputs.validation import *
  File "C:\Users\ksm\WEIS\weis\inputs\validation.py", line 4, in <module>
    from wisdem.inputs import load_yaml, write_yaml, validate_without_defaults, validate_with_defaults, simple_types
ImportError: cannot import name 'validate_without_defaults' from 'wisdem.inputs' (C:\Miniconda3\envs\weis-env\Lib\site-packages\wisdem\inputs\__init__.py)

 non of this are in inputs, load_yaml, write_yaml, validate_without_defaults, validate_with_defaults, simple_types
 they are in validation, however these are not: validate_without_defaults, validate_with_defaults
### Expected behavior
I should get this error 

### Code versions
_List versions only if relevant_
- Python
- …

TheMercer, 2024-07-09:
I have made this ticket on the WISDEM, I didn't realise until I made this post which repo I was looking at

gbarter, 2024-07-09:
Yes, we are aware and resolving this issue as it was a change in WISDEM in the latest release over the weekend.  For the next couple of days, you can use the `fix_win` branch of WEIS and the `fix_scipy` branch of ROSCO.  We are hoping to push those changes to the master branches with the week.

gbarter, 2024-07-09:
Alternatively, you can roll back WISDEM to v3.15

gbarter, 2024-07-16:
(update RAFT too)

TheMercer, 2024-07-16:
Oh okay will try that, thank you

TheMercer, 2024-07-16:
Hey, also. When will this be fixed fully in main builds. So installing weis will "just" work again. Only reason I ask. Is having to balance weis, wisdem, rosco as all separate stuff. Gets hard to die my works security not allowing the build of wisdem or rosco dll etc so it working through pip/conda install. Makes it much much easier and allows Weis to be "centre" hub of the packages

gbarter, 2024-07-16:
We recently pushed the `develop` branch into `master` and will tag a new release.  Pinning WEIS to specific versions of the dependencies, as linked above, will help stabilize things.  A fresh install now should work, as our tests are now passing for the first time in a few months.

TheMercer, 2024-07-16:
Oh okay brilliant, I will check in the morning tomorrow at work. 🤞🤞 

Really appreciate your help.

TheMercer, 2024-07-17:
Master, has got further through. 
```
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox
-----------------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2666, in _call_user_function
    yield
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 264, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 615, in compute
    summary_stats, extreme_table, DELs, Damage, case_list, case_name, chan_time, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 1839, in run_FAST
    comm.send(data, dest=rank_j, tag=0)
  File "mpi4py\MPI\Comm.pyx", line 1406, in mpi4py.MPI.Comm.send
  File "mpi4py\MPI\msgpickle.pxi", line 211, in mpi4py.MPI.PyMPI_send
mpi4py.MPI.Exception: Invalid rank, error stack:
MPI_Send(buf=0x00000137087D6320, count=5611, MPI_BYTE, dest=1, tag=0, MPI_COMM_WORLD) failed
Invalid rank has value 1 but must be nonnegative and less than 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\ksm\WEIS\examples\06_IEA-15-240-RWT\weis_driver_monopile.py", line 16, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ksm\WEIS\weis\glue_code\runWEIS.py", line 213, in run_weis
    wt_opt.run_model()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\problem.py", line 677, in run_model
    self.model.run_solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 4682, in run_solve_nonlinear
    self._solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\group.py", line 3458, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
    ^^^^^^^^^^^^
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\solver.py", line 865, in _gs_iter
    subsys._solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\group.py", line 3458, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
    ^^^^^^^^^^^^
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\solver.py", line 865, in _gs_iter
    subsys._solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 293, in _solve_nonlinear
    self._compute_wrapper()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 247, in _compute_wrapper
    with self._call_user_function('compute'):
  File "C:\Miniconda3\envs\weis-env\Lib\contextlib.py", line 158, in __exit__
    self.gen.throw(value)
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2672, in _call_user_function
    raise err_type(
          ^^^^^^^^^
  File "mpi4py\MPI\ExceptionP.pyx", line 7, in mpi4py.MPI.Exception.__init__
TypeError: an integer is required
```

gbarter, 2024-07-18:
The `mpi4py` package is finicky on Windows and we have removed it from our testing workflow on windows.  Things should still run well without it.

Also, I think we are clearly due for an update on the README install instructions. . .

TheMercer, 2024-07-18:
Oh okay, yeha as when I did the base install. And tried running example 05 15mw, it complains about mpi4py. If I have time tomorrow, I will try grab the error

dzalkind, 2024-07-23:
@gbarter, what kind of instructions should we include?  

I've seen other cases (ROSCO) where Windows users are required to conda install `mpi4py`.

gbarter, 2024-07-25:
Maybe we note in the ReadMe that `mpi4py` will cause some Windows installs to fail but others require it?  Not sure how else to message it and it doesn't seem like we can firmly recommend one approach.

dzalkind, 2024-10-02:
I added a note to this effect in the `docs` branch.  I think the original issue has been resolve, too.  Thanks!

closed 2024-10-02 by dzalkind