NLRWindSystems/WEIS #298: Generating only Cp_Ct_Cq table and new .IN file from WindIO

TheMercer, 2024-07-13:
## Description
Hi, so I am trying to generate a new Cp_Ct_Cq table and .IN from WindIO. 

Closest examples our example 5 & 6. Which generating new OpenFast model, which is great also. However it also always wants to generate BTS files and run LC even tho I get rid of the DLC dict in model.yaml.

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Run example 5 or 6 
2. Remove DLC dict from model.yaml
3. …

### Current behaviour
It will generate the model but then generate BTS files and run DLCs

### Expected behavior
Generate OpenFast model, new Cp_Ct_Cq and .IN, then stop

### Code versions
_List versions only if relevant_
- Python
- …

TheMercer, 2024-07-13:
If you could help, give example or show how I can go from WindIO to OpenFast + Cp_Ct_Cq + .IN without having to make BTS files. Much appreciated

gbarter, 2024-07-16:
Yes, I also see that those OpenFAST files are only generated once TurbSim is done.  I could start a feature branch to fix this issue if that helps.

TheMercer, 2024-07-16:
Yeah, because atleast I could just kill the process after those files are made. 

I think having BTS files first when you don't know the OpenFast is 100% correct yet (due to what ever changes are being made).

Or at somepoint make it possible to have a flag for turbsim files ?

gbarter, 2024-07-16:
I will see what I can do. . .

TheMercer, 2024-07-16:
If you know ish where that lives. I can always see as well and try and help also

TheMercer, 2024-07-16:
So it doesn't "break* other people's runs. You have that flag as true by default. So original functionality is kept. Just can be disabled

dzalkind, 2024-07-16:
There is a `model_only` flag in the modeling options that should skip the simulations: https://github.com/WISDEM/WEIS/blob/77512f2244be4e445425b107c4c40d67326cfaed/weis/inputs/modeling_schema.yaml#L39

https://github.com/WISDEM/WEIS/blob/77512f2244be4e445425b107c4c40d67326cfaed/weis/aeroelasticse/openmdao_openfast.py#L610

TheMercer, 2024-07-16:
Would this skip the BTS generation ? Or could we move bts generation into that ?

TheMercer, 2024-07-16:
I will try that tomorrow, and see if that doesn't generate bts files. (Hopefully not)

dzalkind, 2024-07-16:
Here is where the BTS files are generated: https://github.com/WISDEM/WEIS/blob/77512f2244be4e445425b107c4c40d67326cfaed/weis/aeroelasticse/openmdao_openfast.py#L1821

It should skip that.  My personal workaround would be to only make one short BTS file, [like in this example](https://github.com/WISDEM/WEIS/blob/77512f2244be4e445425b107c4c40d67326cfaed/examples/02_run_openfast_cases/modeling_options.yaml#L39).  That should not take much computing time.

TheMercer, 2024-07-16:
I will try that also, if it does make BTS files

TheMercer, 2024-07-16:
Thank you both for your prompt response.

TheMercer, 2024-07-17:
@dzalkind Model_only to true worked, thank you on that one. 

@gbarter and dzalkind, also highlighted a bug ?

```
Airfoil number 59 tab number 0 has the moment coefficient different between +-180 deg. This is changed to be the same now.
Writing rotor performance text file: C:\Users\ksm\WEIS\examples\06_IEA-15-240-RWT\outputs/06_IEA15_model_only\openfast_runs\rank_0\weis_job_0_Cp_Ct_Cq.txt
Writing new controller parameter file parameter file: C:\Users\ksm\WEIS\examples\06_IEA-15-240-RWT\outputs/06_IEA15_model_only\openfast_runs\rank_0\weis_job_0_DISCON.IN.
Traceback (most recent call last):
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2666, in _call_user_function
    yield
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 264, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 612, in compute
    self.write_FAST(fst_vt, discrete_outputs)
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 2693, in write_FAST
    writer.execute()
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\FAST_writer.py", line 189, in execute
    self.write_MainInput()
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\FAST_writer.py", line 207, in write_MainInput
    f.write('{:<22} {:<11} {:}'.format(self.fst_vt['Fst']['TMax'], 'TMax', '- Total run time (s)\n'))
                                       ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'TMax'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\ksm\WEIS\examples\06_IEA-15-240-RWT\weis_driver_model_only.py", line 16, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ksm\WEIS\weis\glue_code\runWEIS.py", line 213, in run_weis
    wt_opt.run_model()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\problem.py", line 677, in run_model
    self.model.run_solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 4682, in run_solve_nonlinear
    self._solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\group.py", line 3458, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
    ^^^^^^^^^^^^
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\solver.py", line 865, in _gs_iter
    subsys._solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\group.py", line 3458, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
    ^^^^^^^^^^^^
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\nonlinear\nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\solvers\solver.py", line 865, in _gs_iter
    subsys._solve_nonlinear()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 293, in _solve_nonlinear
    self._compute_wrapper()
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 247, in _compute_wrapper
    with self._call_user_function('compute'):
  File "C:\Miniconda3\envs\weis-env\Lib\contextlib.py", line 158, in __exit__
    self.gen.throw(value)
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2672, in _call_user_function
    raise err_type(
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\system.py", line 2666, in _call_user_function
    yield
  File "C:\Miniconda3\envs\weis-env\Lib\site-packages\openmdao\core\explicitcomponent.py", line 264, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 612, in compute
    self.write_FAST(fst_vt, discrete_outputs)
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 2693, in write_FAST
    writer.execute()
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\FAST_writer.py", line 189, in execute
    self.write_MainInput()
  File "C:\Users\ksm\WEIS\weis\aeroelasticse\FAST_writer.py", line 207, in write_MainInput
    f.write('{:<22} {:<11} {:}'.format(self.fst_vt['Fst']['TMax'], 'TMax', '- Total run time (s)\n'))
                                       ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: "'comp.aeroelastic' <class FASTLoadCases>: Error calling compute(), 'TMax'"
```

gbarter, 2024-07-18:
Thank you- we will have to check that one out.

dzalkind, 2024-07-23:
I pushed a fix in #302.

closed 2024-07-28 by TheMercer